espressif/esp-sr #1: 请问怎样添加命令

kaning0760, 2019-08-27:
Commands Recognition Process
Add customized commands to the speech command queue.
Prepare an audio clip of 30 ms (16 KHz, 16 bit, mono).
Input this audio to the MFCC model and get its MFCC features.
Input the obtained MFCC features to MultiNet and get the output phoneme.
Input the obtained phoneme to the Language model and get the output.
Compare the output against the existing speech commands one by one, and output the Command ID of the matching command (if any).
请问是怎样做?用什么文件做输入源?(WAV,MP3...),怎样取MFCC features?

kaning0760, 2019-08-28:
> 你好，你所截取的这一段内容是对 MultiNet 内部的工作流程的介绍。
>
>
> 输入源
>  输入源支持文件（pcm）和实时音频流
>  如果你现在有 ESP32 LyraT_mini 音频开发板，那么可以参考： https://github.com/espressif/esp-skainet/tree/master/examples/get_started 这个示例，是直接输入实时音频流的
>
> 如果你现在没有该开发板，那么针对音频输入部分的驱动需要重新配置或者使用文件作为输入，文件输入可以参考：https://github.com/espressif/esp-sr/blob/master/main/multinet_test.c 这个示例是将 pcm 转存为数组进行输入
>
>
>
> MFCC
>  用户不需要自己编写或处理 MFCC 部分，该部分已经完全编写在 MultiNet 的接口里面，用户只需要传入音频即可，后续的处理无须关心。
>
>
> 更多关于 MultiNet 的使用可以参考： https://github.com/espressif/esp-skaine 这里会包含更多关于 esp-sr 应用层的介绍

你好:
  谢谢你的回复,这个例子Component config 没有 ESP Speech Recognition这一项呢.

----------------------------------------------------------------------------------------------------------------------------------------------------
我们接到了一个语音识别产品的开发任务,要求是离线识别10-15条指令,选了ESP32 LyraT音频开发板.板子买回来几周了只研究出来examples\speech_recognition\asr这个可以识别几条指令.不知道怎样修改指令.
然后也看过esp-sr,可能看得比较肤浅,只是在"menuconfig"添加了几个关键词命令然后就编译了一次,其实我也知道没有这么简单,结果是不能用了.请教一下"离线识别10-15条语音指令"怎样做......?

kaning0760, 2019-08-28:
你好:
  谢谢你的回复,这个例子Component config 没有 ESP Speech Recognition这一项呢.

----------------------------------------------------------------------------------------------------------------------------------------------------
我们接到了一个语音识别产品的开发任务,要求是离线识别10-15条自定义指令,选了ESP32 LyraT音频开发板.板子买回来几周了只研究出来examples\speech_recognition\asr这个可以识别几条指令.不知道怎样修改指令.
然后也看过esp-sr,可能看得比较肤浅,只是在"menuconfig"添加了几个关键词命令然后就编译了一次,其实我也知道没有这么简单,结果是不能用了.请教一下"离线识别10-15条自定义的语音指令"怎样做......?

WeldonWangwang, 2019-08-28:
你好，你所截取的这一段内容是对 MultiNet 内部的工作流程的介绍。
1. 输入源
    输入源支持文件（pcm）和实时音频流
    如果你现在有 ESP32 LyraT_mini 音频开发板，那么可以参考： https://github.com/espressif/esp-skainet/tree/master/examples/get_started 这个示例，是直接输入实时音频流的

    如果你现在没有该开发板，那么针对音频输入部分的驱动需要重新配置或者使用文件作为输入，文件输入可以参考：https://github.com/espressif/esp-sr/blob/master/main/multinet_test.c 这个示例是将 pcm 转存为数组进行输入

2. MFCC
    用户不需要自己编写或处理 MFCC 部分，该部分已经完全编写在 MultiNet 的接口里面，用户只需要传入音频即可，后续的处理无须关心。

更多关于 MultiNet 的使用可以参考： https://github.com/espressif/esp-skaine 这里会包含更多关于 esp-sr 应用层的介绍

WeldonWangwang, 2019-08-29:
你好， 在 https://github.com/espressif/esp-skainet/tree/master/examples/get_started 这个示例下，运行 `make menuconfig`，进入 Component config，即可看到 ESP Speech Recognition这一项，如果使用的是 ESP-IDF V3.2，那么在 Component config 中第 18 行

kaning0760, 2019-08-29:
你好,可能ESP-IDF问题吧,真的没有Component config选项.正在下载一个新.可是下了一天,失败了很多次,下载速度只有10K-8K.现在还没能下载下来.......这个资源国内有下吗?之前光开发环境弄了一个星期,国内的限制太多了,下载也是很奢侈的事.

kaning0760, 2019-08-30:
你好,我重新下载了一个IDF安装了一下

应该安装是没问题的.

按你说的重新下载了一个esp-skainet包,打开/examples/get_started例子,运行 make menuconfig，进入 Component config.上图已经是全部的选项,还是没有ESP Speech Recognition这一项.求教.......

kaning0760, 2019-08-30:
你好,可不可以发我一个你编译过可以自定义语音识别关键词的党演示工程包,非常感谢!
我重新下载了一个IDF安装了一下

应该安装是没问题的.

按你说的重新下载了一个esp-skainet包,打开/examples/get_started例子,运行 make menuconfig，进入 Component config.上图已经是全部的选项,还是没有ESP Speech Recognition这一项.求教.......

closed 2019-10-09 by feizi