espressif/esp-sr #2: ESP Speech Recognition没有这一项

kaning0760, 2019-09-02:
在 https://github.com/espressif/esp-skainet/tree/master/examples/get_started 这个示例下，运行 make menuconfig，进入 Component config，
[image]

kaning0760, 2019-09-02:
可不可以发我一个你编译过可以自定义语音识别关键词的党演示工程包,非常感谢!

WeldonWangwang, 2019-09-03:
你好，这应该是没有更新 submodule 导致，建议运行：`git submodule update --init` 后再次尝试下！

kaning0760, 2019-09-03:
[image]
你好,之前已经更新 submodule了,不过我再做了一次,还是不行,还是没有啊!

kaning0760, 2019-09-03:
[image]
[image]

feizi, 2019-09-03:
是在 esp-skainet工程目录下运行 git submodule update --init
运行完，确保在你的esp-skainet工程里，有如下文件夹
 /esp-skainet/components/esp-sr

kaning0760, 2019-09-03:
好像有点问题!

kaning0760, 2019-09-03:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6664
load:0x40078000,len:11800
load:0x40080400,len:6736
entry 0x40080764
[0;32mI (29) boot: ESP-IDF v4.1-dev-104-gaa087667d-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 20:20:59[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) boot: SPI Speed      : 80MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 factory          factory app      00 00 00010000 003c0000[0m
[0;32mI (66) boot:  1 nvs              WiFi data        01 02 003d0000 00004000[0m
[0;32mI (74) boot: End of partition table[0m
[0;32mI (78) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f503c (2052156) map[0m
[0;32mI (684) esp_image: segment 1: paddr=0x00205064 vaddr=0x3ffb0000 size=0x028ac ( 10412) load[0m
[0;32mI (688) esp_image: segment 2: paddr=0x00207918 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (691) esp_image: segment 3: paddr=0x00207d20 vaddr=0x40080400 size=0x082f0 ( 33520) load[0m
[0;32mI (712) esp_image: segment 4: paddr=0x00210018 vaddr=0x400d0018 size=0x514c0 (332992) map[0m
[0;32mI (809) esp_image: segment 5: paddr=0x002614e0 vaddr=0x400886f0 size=0x07674 ( 30324) load[0m
[0;32mI (830) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (830) boot: Disabling RNG early entropy source...[0m
[0;32mI (830) psram: This chip is ESP32-D0WD[0m
[0;32mI (835) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (839) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (845) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (852) cpu_start: Pro cpu up.[0m
[0;32mI (856) cpu_start: Application information:[0m
[0;32mI (860) cpu_start: Project name:     esp_sr_public[0m
[0;32mI (866) cpu_start: App version:      v0.1-4-gc7167de-dirty[0m
[0;32mI (872) cpu_start: Compile time:     Sep  3 2019 20:20:23[0m
[0;32mI (878) cpu_start: ELF file SHA256:  698885d4c53f2bfe...[0m
[0;32mI (884) cpu_start: ESP-IDF:          v4.1-dev-104-gaa087667d-dirty[0m
[0;32mI (891) cpu_start: Starting app cpu, entry point is 0x40081488[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1391) spiram: SPI SRAM memory test OK[0m
[0;32mI (1391) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1392) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1398) heap_init: At 3FFB4A70 len 0002B590 (173 KiB): DRAM[0m
[0;32mI (1404) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1411) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1417) heap_init: At 4008FD64 len 0001029C (64 KiB): IRAM[0m
[0;32mI (1423) cpu_start: Pro cpu start user code[0m
[0;32mI (1428) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1449) spi_flash: detected chip: generic[0m
[0;32mI (1449) spi_flash: flash io: dio[0m
[0;32mI (1449) cpu_start: Chip Revision: 1[0m
[0;33mW (1452) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.[0m
[0;32mI (1462) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1472) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
Start free RAM size: 4486664
Quantized wakeNet5: wakeNet5X2_v2_hilexin_5_0.98_0.90, mode:0 (Aug 21 2019 14:25:59)
WakeNet RAM size: 9516
RAM size after WakeNet init: 4477148
WN test successfully, 1.41: Neural network detection triggered output 1.
Done! Took 1732 ms to parse 3810 ms worth of samples in 127 iterations. CPU loading(single core):45.5%
TEST1 FINISHED

Start free RAM size: 4468892
Quantized MN1 (Aug 19 2019 16:38:49)
[0;32mI (4492) MN: ---------------------SPEECH COMMANDS---------------------[0m
[0;32mI (4492) MN: Command ID0, phrase 0: da kai kong tiao[0m
[0;32mI (4492) MN: Command ID1, phrase 1: guan bi kong tiao[0m
[0;32mI (4502) MN: Command ID2, phrase 2: zeng da feng su[0m
[0;32mI (4512) MN: Command ID3, phrase 3: jian xiao feng su[0m
[0;32mI (4512) MN: Command ID4, phrase 4: sheng gao yi du[0m
[0;32mI (4522) MN: Command ID5, phrase 5: jiang di yi du[0m
[0;32mI (4522) MN: Command ID6, phrase 6: zhi re mo shi[0m
[0;32mI (4532) MN: Command ID7, phrase 7: zhi leng mo shi[0m
[0;32mI (4532) MN: Command ID8, phrase 8: song feng mo shi[0m
[0;32mI (4542) MN: Command ID9, phrase 9: jie neng mo shi[0m
[0;32mI (4552) MN: Command ID10, phrase 10: guan bi jie neng mo shi[0m
[0;32mI (4552) MN: Command ID11, phrase 11: chu shi mo shi[0m
[0;32mI (4562) MN: Command ID12, phrase 12: guan bi chu shi mo shi[0m
[0;32mI (4562) MN: Command ID13, phrase 13: da kai lan ya[0m
[0;32mI (4572) MN: Command ID14, phrase 14: guan bi lan ya[0m
[0;32mI (4572) MN: Command ID15, phrase 15: bo fang ge qu[0m
[0;32mI (4582) MN: Command ID16, phrase 16: zan ting bo fang[0m
[0;32mI (4592) MN: Command ID17, phrase 17: ding shi yi xiao shi[0m
[0;32mI (4592) MN: Command ID18, phrase 18: da kai dian deng[0m
[0;32mI (4602) MN: Command ID19, phrase 19: guan bi dian deng[0m
[0;32mI (4612) MN: ---------------------------------------------------------
[0m
multinet RAM size: 53140
RAM size after multinet init: 4415752
phrase ID: 18, prob: 0.866630
MN test successfully, Commands ID: 18.
TEST2 FINISHED

NS test successfully

AGC test successfully

AEC test successfully

VAD test successfully

TEST3 FINISHED

kaning0760, 2019-09-03:
工程是~/esp/esp-skainet/components/esp-sr,编译后烧到板上,以上是打印出来的信息.我对咪说"hi,乐鑫"没反应.然后说那20个指命也没反应.

closed 2019-10-09 by feizi