espressif/esp-sr #3: Open Source Speech Recognition Training

maxsu, 2019-10-03:
I'd like to make a case for Espressif to open source their training code for ESP wake word programming, and open their neural net distillation toolchain for ESP32.

This will accomplish two salient business goals:

1. Take attention away from Microsoft's [EdgeML](https://github.com/microsoft/EdgeML) / FastGRNN, which is the only other open wake-word solution capable of targeting ESP32/Cortex M0+ class microcontrollers.

2. Provide a ESP32 specific alternative to the ST32 with ST's [X-CUBE-AI](https://www.st.com/resource/en/data_brief/x-cube-ai.pdf).

The risk: this could reduce keyword consulting opportunities for Espressif.
However:
- technically unsophisticated customers will still require assistance
- this will attract technical and nontechnical customers to ESP32s

Case Study: [This paper](https://nmbu.brage.unit.no/nmbu-xmlui/bitstream/handle/11250/2611624/report-print1.pdf?sequence=3),  implements an environmental noise detector AI in a ST microsystems MCU. They adapt a FastGRNN keyword detector and compile it with ST's X-CODE-AI system to target an ST Cortex M0+.

Why not make ESP-32 a primary choice for this kind of work? I hope you will agree that Opening ESP-SR moves you in that direction.

feizi, 2019-10-09:
Thanks for your suggestion.
Developing a neural net distillation toolchain for ESP32 is a good idea. As far as I know, some guys plan to do this.

maxsu, 2019-10-09:
@feizi - that sounds exciting! I hope they also include a wake word trainer! Can you recall their names?

I will close the current issue as it does not seem like something immediately actionable.

closed 2019-10-09 by maxsu