espressif/esp-sr #7: c++ wakeword example

Romkabouter, 2019-11-24:
Hi, 

Is it possible to provide an example using c++?

I have created a c++ program which is calling the wakeNet.
This compiles and runs ok, but when I call:
int r = wakenet->detect(model_data, buffer);
I get this error message:

Fail: index is out of range, the min index is 1, the max index is 1

feizi, 2019-11-25:
That error message may appear in function `set_det_threshold`, `get_det_threshould`,`get_word_name`, instead of `detect` when parameter `index` value is error.    
So please check the parameter `index` first.

Romkabouter, 2019-11-25:
I am using the exact Task as used in https://github.com/espressif/esp-sr/blob/master/main/wakenet_test.c

What is the index parameter in that testcase?
The call  int frequency = wakenet->get_samp_rate(model_data); works as expected.

When I comment out the detect call, I do not get an error message

feizi, 2019-11-26:
It is strange, I will test wakenet by C++.

Romkabouter, 2019-11-28:
Additional request that if it is not an issue, a sample project can be pushed to github?
Thanks in advance

Romkabouter, 2019-12-12:
Hi feizi, did you already have time to look into this issue?

feizi, 2019-12-23:
I do not get the above error message when I test wakenet by C++.

1. modify the main/main.c to main/main.cpp, 
2. modify main function:
```
extern "C" {
void app_main()
{
    // test wakenet
    wakenet_test();
    vTaskDelay(3000 / portTICK_PERIOD_MS);

}
}
```
3. modify ./main/include/wakenet_test.h
```
#pragma once
#ifdef _cpulspuls
extern "C" {
#endif

void wakenet_test();

#ifdef _cplusplus
}
#endif
```
4. make flash

closed 2019-12-27 by feizi

Romkabouter, 2019-12-27:
Thank you, I have got the test working now. 
I am dutch and I do not speak Chinese, is it possible to include a wav file from the hilexin.h in order to hear how the wakeword should be spoken?