espressif/esp-sr #14: Feature request: In speech recognition, pass detected phoneme to callbackfunction

bluelu, 2020-05-26:
Hi,

For the speech detection, would it be possible to pass each detected phoneme (together with their probabilities?) to a callback function, which is triggered for each processed frame?

One could then stream those phonemes to a server which could do a more advanced command recognition.

At the moment, all the commands have to be predefined which is more focussed on stand alone devices.  

Not sure though if this is possible though technically since it's a binary library currently.

Thanks

feizi, 2020-05-26:
In fact, if one can upload data to the server, I suggest he streams the microphone data to server directly.

bluelu, 2020-05-26:
That was my backup plan as well. I will do it like that then. Thank you.

closed 2020-06-22 by feizi