espressif/esp-sr #19: 官方公开的唤醒词不够人性化

1003657663, 2020-11-20:
从已经公开的唤醒词中看出，唤醒词总是从，“Hi”开始。
实际上人说话都是直接叫名字，特别是中文，很少说，Hi，谁谁谁。而且在家里控制智能家电，总是说，“Hi”，显得很刻意，很奇怪。这也是为什么“小爱同学”和“天猫精灵”直接用名字做唤醒词的原因。

希望官方能提供一些更合理的公开唤醒词，不加“Hi”，“Ok”等词汇。比如大家都希望拥有的
“贾维斯”
“初音未来”  这种，对于开发者比较友好的，其他厂商又不会用的词。千万不要“Hi，贾维斯”这样。

真诚的期待！

feizi, 2020-11-24:
唤醒词的训练需要比较多的 **专门录制的唤醒词语料**，不同开发者需求不同。  
如果开发者可以提供语料，我们可以帮忙训练模型。
如果开发者没有语料，但有定制的需求，可以和我们公司商务联系，参考[唤醒词定制](https://github.com/espressif/esp-sr/blob/master/wake_word_engine/%E4%B9%90%E9%91%AB%E8%AF%AD%E9%9F%B3%E5%94%A4%E9%86%92%E8%AF%8D%E5%AE%9A%E5%88%B6%E6%B5%81%E7%A8%8B.md)

anuradhawick, 2021-02-26:
Hi @feizi 
If I want to train everything by myself, how can I do that? Could you please advise me?

feizi, 2021-05-10:
Sorry for the late reply.
1. you need  to train a NN model. [ML-KWS-for-MCU](https://github.com/ARM-software/ML-KWS-for-MCU) is a good project you can refer to 
2. you need to develop the conv or other NN API for your model. （we plan to open some common APIs, such as conv1d, conv2d, but it has not been completed yet）

closed 2021-05-10 by feizi