espressif/esp-sr #25: Change speech commands on the fly?

expresspotato, 2021-07-16:
Hello,

Does anyone know how to change recognized commands on the fly? It looks like there are functions to "reset" the command, but not exactly sure what this does: `reset_speech_commands(model_iface_data_t *model_data, char* command_str, char *err_phrase_id)`. I tried to call it but got redefinition warnings when including mn_process_commands.h in main.c.

Found in:
https://github.com/espressif/esp-sr/blob/master/speech_command_recognition/mn_process_commands.c

Also there appears to be some additional functions to "Support Web Server" but there isn't any documentation on it.

https://github.com/espressif/esp-sr/commit/9159350d04ef468c0d264a4adc739aa5adaecfc8

expresspotato, 2021-09-19:
Hi Fezi, any update on this?

feizi, 2021-11-08:
@expresspotato , Now we have fixed the bug of reset API. You can use this API to change speech commands on the fly.
Please refer to [README](https://github.com/espressif/esp-sr/tree/master/docs/speech_command_recognition)

expresspotato, 2021-11-10:
Well doesn't really help me... Commit https://github.com/espressif/esp-sr/commit/20915ded8b351f9536b138a18a48646314fb080f pretty much broke all English support on the ESP32 (LyraT v4.3)... 

And this removed the Hi Jeson, wakeup word... https://github.com/espressif/esp-sr/commit/a78b3d22b18446383328749e09d140161ae9bddd#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20a

Is there any material reason why the ESP32-S3 is needed? The specs seem quite similar and there's a lack of availability of the dev boards, namely the ESP32-S3-Korvo-2 is a real problem.

feizi, 2021-11-10:
Compared with ESP32, ESP32-S3 supports AI acceleration instructions and a faster SPI-RAM, which can run about 4~8 times larger neural network models.   
At the same time, we developed a powerful Audio Front End based on ESP32-S3 instructions. 

The English multinet5(ESP32-S3) performance has been greatly improved, so we do not support the old English multinet.

We use Hi ESP to replace Hi Jeson as the English wake word for Espressif. (currently only support ESP32-S3) 

ESP32-S3-KorvoV4.0 will be released this month.

feizi, 2021-11-10:
@expresspotato, I guess you might be interested in [esp-box](ESP-BOXhttps://github.com/espressif/esp-box)

closed 2021-12-10 by feizi