espressif/esp-sr #27: audio_front_end: Wrong frame size for fetch?

expresspotato, 2021-07-31:
It look like the wrong frame size is being returned in the audio_front_end component. For get_fetch_chunksize, it's returning 512 with no wakenet included, or 480 with Wakenet5X3. Strange values for an input of only 256x2 = 512 interlaced samples given that the output is 1 channel. I would have expected a fetch chunk size of 256.

get_feed_chunksize returns: 256
get_fetch_chunksize returns (with no wakenet set) -> 512
get_fetch_chunksize returns (with wakenet set ) -> 480

https://github.com/espressif/esp-sr/blob/7c8e4ca09d8298c60bc7c9c8b25747879cbf9ff0/audio_front_end/esp_afe_sr_1mic.c#L209

feizi, 2021-08-09:
If wakenet has been set, get_fetch_chunksize return the chunksize of specific wakenet(512(32ms) for wakenet7, 480(30ms) for wakenet5)

expresspotato, 2021-08-18:
Hmm, do you have an example of using this with the audio pipeline? I'm not seeing these same results.

If input is 2 channels, I would expect it to have a feed chunk size (512) > fetch chunk size (256)

expresspotato, 2021-09-19:
This can be closed...
It should be noted that you need to feed twice, for each call to fetch if feed chunk size is 256 and fetch chunk size is 512.

smallbraingames, 2024-10-24:
> Please refer to [example ](https://github.com/espressif/esp-skainet/tree/esp32s3_dev/examples/online_test)of esp-skainet

This example is 404ing, does it still exist?

In general, I'm looking for approaches to sync data from speaker and mic