espressif/esp-sr #30: How to support the recognition of the data input by the digital microphone I2S?

Kevincoooool, 2021-11-20:
Hello, I am using a digital microphone, the model is MSM261S4030H0R, when I use the old version of esp-sr, after I get the i2s data, perform the following operations on the data and then send the processed data to the multinet, 
` 
i2s_read(I2S_NUM, samp, samp_len, &read_len, portMAX_DELAY);
            for (int x = 0; x < cfg->item_size / 4; x++)
            {
                int s1 = ((samp[x * 4] + samp[x * 4 + 1]) >> 13) & 0x0000FFFF;
                int s2 = ((samp[x * 4 + 2] + samp[x * 4 + 3]) << 3) & 0xFFFF0000;
                samp[x] = s1 | s2;
            }
            xQueueSend(*cfg->queue, samp, portMAX_DELAY);
`
it can be recognized normally, but now it is used After the new version of esp-sr can not be recognized, what is the problem, thank you!

feizi, 2021-11-29:
Which multinet are you using?

Kevincoooool, 2021-11-29:
libwakenet.a

feizi, 2021-11-30:
When you create multinet, it will print the version of multinet, like this:
```
Quantized8 Multinet5:  MN5Q8_v1_english_8_0.9_0.90 
```
which version of multinet are you using?

Kevincoooool, 2021-11-30:
when i use 
`Quantized wakeNet5: wakeNet5_v1_hilexin_5_0.95_0.90, mode:1 (Oct 14 2020 16:26:17)
wake word number = 1, word1 name = hilexin
SINGLE_RECOGNITION: 2_0 MN1_4; core: 0; (May 15 2020 14:50:27)
SHIFT: 8, 12, 17, 17, 19, 17, 6, 16, 15, 14,`
Everything works fine，But this can't work:
`accelate model: 1
MC Quantized wakeNet7: wakeNet7_v1_xiaoaitongxue_6_0.97_0.92, mode:2, p:3, (Sep 24 2021 18:25:10)
Initial TWO-MIC auido front-end for speech recognition, mode:0, (Sep 24 2021 20:54:19)
------------detect start------------`
I use a digital microphone of a single microphone, and read the data directly through I2S，

feizi, 2021-12-02:
Wakenet7 needs ESP32-S3 to run in real time

Kevincoooool, 2021-12-02:
> Wakenet7 needs ESP32-S3 to run in real time

Yes I know,I'm using ESP32-S3 D3N8R8

Kevincoooool, 2021-12-04:
> Wakenet7 needs ESP32-S3 to run in real time

Hello, can you please provide a detailed document, a document on how to convert input data for different microphones (ADC\I2S\MEMS) for identification

feizi, 2021-12-10:
Whether it is a digital microphone or an analog microphone, the input of wakenet is the same. 
You need to pay attention to the number of microphones and the wake-up channels.
I suggest you use [Audio Front End](https://github.com/espressif/esp-sr/tree/master/docs/audio_front_end). That document defines the input format in detail.

Kevincoooool, 2021-12-23:
> Whether it is a digital microphone or an analog microphone, the input of wakenet is the same. You need to pay attention to the number of microphones and the wake-up channels. I suggest you use [Audio Front End](https://github.com/espressif/esp-sr/tree/master/docs/audio_front_end) which includes wakenet and speech enhancement. That document defines the input format in detail.

Ok, now the board I made by myself is the same as the situation of ESP32 S3-EYE. They are all MEMS digital microphones. Can you provide a demo of the single microphone identification of ESP32 S3-EYE?

feizi, 2021-12-30:
We will support ESP32 S3-EYE in esp-skainet.

WeldonWangwang, 2022-01-10:
Hi, @Kevincoooool
We support the digital microphone I2S in [esp-skainet](https://github.com/espressif/esp-skainet/tree/master/examples/wake_word_detection) , you can choose ESP-EYE board(mic: MSM261S4030H0R) to test this example.
If you use other microphones, please refer this [code](https://github.com/espressif/esp-skainet/tree/master/components/hardware_driver/boards/esp32s3-eye) .
Thanks!

Kevincoooool, 2022-01-10:
> [Audio Front End](https://github.com/espressif/esp-sr/tree/master/docs/audio_front_end)

Yes, I know this example, but the code of this example only supports the recognition of the old version of esp-sr. When I use the master branch of esp-sr, it cannot be recognized normally with the original code. I use esp32s3, which can be used at present The old version of esp-sr can be recognized, and the new version such as Audio Front End cannot be recognized by my previous code.

Kevincoooool, 2022-01-10:
> Hi, @Kevincoooool We support the digital microphone I2S in [esp-skainet](https://github.com/espressif/esp-skainet/tree/master/examples/wake_word_detection) , you can choose ESP-EYE board(mic: [...]
Thank you very much！！！

Kevincoooool, 2022-01-10:
> Hi, @Kevincoooool We support the digital microphone I2S in [esp-skainet](https://github.com/espressif/esp-skainet/tree/master/examples/wake_word_detection) , you can choose ESP-EYE board(mic: [...]

I tested it, but I still can't use the digital microphone to identify it. I used the configuration of ESP32S3-EYE to compile and download it.

WeldonWangwang, 2022-01-10:
@Kevincoooool 
1. 请问你使用的是哪款硬件，如果是自己设计的硬件，首先请确保 mic 正常工作，建议将 mic 采集到的音频进行存储，查看是否正常
2. 如果使用的是 ESP32S3-EYE，麻烦提供一下 log

StuartIanNaylor, 2022-02-24:
Currently the AEC is fed by a 3rd ADC loopback channel from the DAC as the algs need to be tightly synced.
You prob could but its hard to envisage how with the current pipeline for 2 mics -> AEC -> BSS.
A product has turned up ESP32-S3-Box-Lite which has only a 2 channel ADC and it will be interesting to see what processing pipeline is provided.
I am hoping with a fixed frequency RTOS that the loopback can be dropped with a circular buffer to compensate for latency of the ref signal... ?

Kevincoooool, 2023-02-24:
> @Kevincoooool
> 
> 1. 请问你使用的是哪款硬件，如果是自己设计的硬件，首先请确保 mic 正常工作，建议将 mic 采集到的音频进行存储，查看是否正常
> 2. 如果使用的是 ESP32S3-EYE，麻烦提供一下 log

您好，我发现ESP32S3-EYE的MIC声道脚是接地的，所以数据是左声道，提取音频数据用这个代码:
` int32_t *tmp_buff = buffer;
    for (int i = 0; i < audio_chunksize; i++) {
        tmp_buff[i] = tmp_buff[i] >> 14; // 32:8为有效位， 8:0为低8位， 全为0， AFE的输入为16位语音数据，拿29：13位是为了对语音信号放大。
    }`
而我自己的板子声道脚是接的3.3V，所以导致我接受到的数据是右声道，提取右声道应该取哪些位呢？谢谢

closed 2023-06-14 by feizi