espressif/esp-sr #31: Syntax definition for commands

obones, 2021-12-20:
Hello,

On this page, we see the default commands for Chinese or English: https://github.com/espressif/esp-sr/blob/master/docs/speech_command_recognition/README.md#1menuconfig-before-compilation

The letters in there seem very cryptic, could you please document what the syntax is meant to be?

feizi, 2021-12-27:
For Chinese, the letters is Pinyin without tone.
For English , the letter is International Phonetic Alphabet. You can check [multinet_g2p.py](https://github.com/espressif/esp-sr/blob/master/tool/multinet_g2p.py) to get the map.
`
alphabet={"AE1": "a", "N": "N", " ": " ", "OW1": "b", "V": "V", "AH0": "c", "L": "L", "F": "F", "EY1": "d", "S": "S", "B": "B", "R": "R", "AO1": "e", "D": "D", "AH1": "c", "EH1": "f", "OW0": "b", "IH0": "g", "G": "G", "HH": "h", "K": "K", "IH1": "g", "W": "W", "AY1": "i", "T": "T", "M": "M", "Z": "Z", "DH": "j", "ER0": "k", "P": "P", "NG": "l", "IY1": "m", "AA1": "n", "Y": "Y", "UW1": "o", "IY0": "m", "EH2": "f", "CH": "p", "AE0": "a", "JH": "q", "ZH": "r", "AA2": "n", "SH": "s", "AW1": "t", "OY1": "u", "AW2": "t", "IH2": "g", "AE2": "a", "EY2": "d", "ER1": "k", "TH": "v", "UH1": "w", "UW2": "o", "OW2": "b", "AY2": "i", "UW0": "o", "AH2": "c", "EH0": "f", "AW0": "t", "AO2": "e", "AO0": "e", "UH0": "w", "UH2": "w", "AA0": "n", "AY0": "i", "IY2": "m", "EY0": "d", "ER2": "k", "OY2": "u", "OY0": "u"}
`

closed 2021-12-28 by feizi

obones, 2022-01-05:
Thanks, but IPA uses unicode characters where you are using ASCII characters. I believe it's one of the ASCII encodings for IPA, but which one? Is it one of these : https://en.wikipedia.org/wiki/Comparison_of_ASCII_encodings_of_the_International_Phonetic_Alphabet

feizi, 2022-01-06:
Yes， It is a ASCII encodings for IPA ddefined by ourselves.