espressif/esp-sr #35: AEC handle failed to initialize for esp32 target

keeofkoo, 2022-03-08:
I am trying to hook up the algorithm stream with an audio pipeline on my `esp32`-based audio board, hoping to enhance the audio quality with the AEC algorithm and everything. 

However, the app crashes where `algo->afe_handle` is [being created](https://github.com/espressif/esp-adf/blob/2dffa85dc5183db1bfe0f9808c0ed61a3194e0c6/components/audio_stream/algorithm_stream.c#L133). 

After some debugging and investigation, I found that the problem might be the [misconfigured](https://github.com/espressif/esp-sr/blob/d05cf97972c0a324d331437d6a37229a2d96f3cf/src/esp_afe_sr_1mic.ref#L114) `enable_psram` parameter, which is always `0` regardless of what `afe_config->alloc_from_psram` might be.

Should this be a bug, a fix would be very much appreciated.

Kee

aihun1989, 2022-03-15:
Hi Kee,

    Which branch of esp-sr do you use for testing, master? or release/v1.0?

keeofkoo, 2022-03-16:
`espressif/esp-sr@83e2176`, as referenced by `espressif/esp-adf@master`.

aihun1989, 2022-03-17:
Hi keeofkoo,
     We test the espressif/esp-sr@83e2176 at branch release/v1.0 is ok. The afe_config->alloc_from_psram is not only for AEC. The enable_psram, if you mean it's the parameter of AEC init function. It's always 0 on esp32 target platform is ok. So it's not the root cause of fail.
     Maybe there is something wrong with your configuration or function call. You can post your configuration of `afe_config_t afe_config` . And how to use the API of esp-sr. You can refer the examples/wake_word_detection/main/main.c of espressif/esp-skainet on master branch.

keeofkoo, 2022-04-08:
@aihun1989, I'm bringing some updates on the issue.

I have tried `examples/wake_word_detection` of `espressif/esp-skainet`, which works, but I don't see there's much to do with the problem I have encountered.

In order to "reveal" the problem, I stripped all but `afe_handle->create_from_config(&afe_config)` off the example, trying to figure out the conditions where the process would crash. Before the "process", I allocated some memory from the internal heap to limit the free internal heap size. The `SPIRAM` was enabled, and the same `AFE` configuration was used. Through the experiment, I found:

* For `esp32` target, the "process" would crash when the free internal heap size was less than roughly 80kB.
* For `esp32s3` target, the "process" would not crash even when there was only 10kB of free internal heap memory.

It's easy to see that, for `esp32` target, `AFE` requires way more internal heap memory (80kB is quite a lot) than `esp32s3` target.

FYI, the code that I used for experiment:

```c
#include "esp_afe_sr_iface.h"
#include "esp_afe_sr_models.h"
#include "esp_log.h"

static const char *TAG = "AFE_TEST";

static const esp_afe_sr_iface_t *afe_handle = NULL;

static inline size_t heap_internal(void) {
    return heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL);
}

static inline size_t heap_external(void) {
    return heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM);
}

void app_main() {
#if CONFIG_IDF_TARGET_ESP32 || defined CONFIG_ESP32_S3_EYE_BOARD
    afe_handle = &esp_afe_sr_1mic;
#else
    afe_handle = &esp_afe_sr_2mic;
#endif
    ESP_LOGI(TAG, "heap(1): %d/%d", heap_internal(), heap_external());
    do {
        size_t bank_num = 100; // Change this to emulate different conditions
        void *bank = NULL;
        for (int i = 0; i < bank_num; i++) {
            bank = heap_caps_malloc(1024, MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL);
            if (bank == NULL) {
                abort();
            }
        }
    } while (false);
    ESP_LOGI(TAG, "heap(2): %d/%d", heap_internal(), heap_external());
    afe_config_t afe_config = AFE_CONFIG_DEFAULT();
    afe_config.vad_init = false;
    afe_config.wakenet_init = false;
    afe_config.afe_mode = SR_MODE_HIGH_PERF;
    afe_config.alloc_from_psram = AFE_PSRAM_HIGH_COST;
    esp_afe_sr_data_t *afe_data = afe_handle->create_from_config(&afe_config);
    ESP_LOGI(TAG, "heap(3): %d/%d", heap_internal(), heap_external());
}
```

BlueSkyB, 2022-04-24:
@keeofkoo 
Yes, due to different implementation methods of algorithm, the memory occupation is different on esp32 and esp32s3.
On the esp32 platform, the memory of AEC/NS algorithm is all in internal memory. If both are enabled, at least 150kb of internal memory needs to be reserved.
On the esp32s3 platform, because the algorithm can use psram according to the mode of alloc_from_psram, so it can occupy less internal memory.

keeofkoo, 2022-04-24:
@BlueSkyB Thanks for the clarification.

closed 2022-04-24 by keeofkoo