espressif/esp-sr #38: Can ESP provide some generic wake words？

balanceTWK, 2022-04-06:


balanceTWK, 2022-04-06:
Or offer some ways to train wake words.

feizi, 2022-04-08:
We plan to collect more corpus in order to provide more wake words. Which words do you think are more generic wake words?

balanceTWK, 2022-04-11:
> We plan to collect more corpus in order to provide more wake words. Which words do you think are generic wake words?

@feizi 

Chinese： "智能管家"；“语音助手”；“AI 管家”；such non-personalized wake words.

sohotcall, 2023-07-18:
@feizi
English: "computer"

feizi, 2023-08-15:
Hi @sohotcall , We are reducing the cost of wake word training. When we finish this, we will post more wake words for free.

mike-2020, 2023-09-17:
Most of the commercial ASR chips/modules support wakeup word custermization at build/compile time. Some of them even supports wakeup words self-learning at runtime. 

Does ESP-SR have plan to support similar capability?

feizi, 2023-09-18:
> Most of the commercial ASR chips/modules support wakeup word custermization at build/compile time. Some of them even supports wakeup words self-learning at runtime.

We have to make a choice between a very low-memory KWS and a customizable KWS. Currently, wakenet9(our latest wake model) parameter is about 250KB. It is difficult for this model to customize a new wake word using a few samples.

Now, our solution is to generate a large number of samples by TTS to customize the wake word. For community developers, we will collect some favorite wake words and regularly release these wake word models based on our capabilities. These models can be used commercially for free.

feizi, 2023-12-14:
Now we can provide more generic wake words by TTS samples.

feizi, 2023-12-14:
Hi all, now we can provide some generic wake words by TTS samples.

closed 2023-12-14 by feizi