espressif/esp-sr #43: undefined reference to `get_model_base_path'

chengcharchar, 2022-08-21:
为什么我在编译的时候会报这个错？大家有遇到吗？用VScode IDF4.4 esp-adf2.4 编译
`
d:/environment/esp-idf/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Environment/ESP-IDF/Espressif/esp-adf/components/esp-sr/lib/esp32s3/libwakenet.a(wakenet9_quantized.c.obj): in function `model_create':
/home/sunxiangyu/aishare/master/esp_sr_lib/build/../components/wakenet/wakenet9_quantized.c:486: undefined reference to `get_model_base_path'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
`

ericnts, 2022-08-23:
快点来人解决一下~~~~

[image]

liminghui-sandra, 2022-08-23:
> 快点来人解决一下~~~~
> 
> [image]

请更新 submodule:
git submodule update --init --recursive

ericnts, 2022-08-24:
> git submodule update --init --recursive

并没有作用，我是新down的adf

feizi, 2022-08-29:
This issue only occurs on MacOS systems, which has been solved by modifying CMakeList.txt