espressif/esp-sr #48: How to online switching of Chinese and English Speech Command Recognition?

casitelwithdean, 2022-10-15:
I am using esp32-s3，is there a way to  switching of Chinese and English Speech Command Recognition online ?Thanks!

feizi, 2022-10-17:
Yes, you can refer to [esp-box](https://github.com/espressif/esp-box/tree/master/examples/factory_demo) to switch speech commands recognition model online.   
The basic steps is as follows:  
1. Select Enghlish and Chinese model in sdkconfig to load
2. Use `char *mn_name = esp_srmodel_filter(models, ESP_MN_PREFIX, ESP_MN_ENGLISH);` or `char *mn_name = esp_srmodel_filter(models, ESP_MN_PREFIX, ESP_MN_ENGLISH);` to get English or Chinese model name
3. Get multinet handle from model name by `esp_mn_iface_t *multinet = esp_mn_handle_from_name(mn_name);`
4. Create, Destroy or Detect by multinet handle and model name

casitelwithdean, 2022-10-17:
> Yes, you can refer to [esp-box](https://github.com/espressif/esp-box/tree/master/examples/factory_demo) to switch speech commands recognition model online. The basic steps is as follows:
> 
> 1. Select Enghlish and Chinese model in sdkconfig to load
> 2. Use `char *mn_name = esp_srmodel_filter(models, ESP_MN_PREFIX, ESP_MN_ENGLISH);` or `char *mn_name = esp_srmodel_filter(models, ESP_MN_PREFIX, ESP_MN_CHINESE);` to get English or Chinese model name
> 3. Get multinet handle from model name by `esp_mn_iface_t *multinet = esp_mn_handle_from_name(mn_name);`
> 4. Create, Destroy or Detect by multinet handle and model name

thank u!

closed 2023-03-09 by casitelwithdean