espressif/esp-sr #50: 请问0.8.0版本如何配置为连续识别(CONTINUOUS_RECOGNITION)模式

wangxiankang, 2022-10-20:
您好，我在esp-sr ## 0.7.0 可以通过以下配置修改为连续识别模式，升级到 ## 0.8.0后发现Kconfig文件改动了很多，没看到在哪里设置连续识别模式，请相关开发大佬帮忙解答以下，0.8.0版本的sr如何配置成连续识别模式，我的开发板是ESP32_S3_KORVO2_V3，测试demo为wwe.

### 命令词识别模式

命令词识别支持两种基本模式：
- SINGLE_RECOGNITION 模式

 即单次识别模式，当使用该模式时，用户在进行命令词识别时，必须将单独的单个命令词短语音频送入 MultiNet。  
 比如在唤醒后说出：打开电灯。则 MultiNet 会识别成功并返回对应的 Command ID。如果识别失败，必须等 sample_length 时长结束后才能进行下次识别。  
 当配合唤醒使用时，如果用户在唤醒后只需要识别一个关键字返回即可，推荐使用该模式。  
 
- CONTINUOUS_RECOGNITION 模式

 即连续识别模式，当使用该模式时，用户可以将多个命令词连续送入 MultiNet。  
 比如在唤醒后，可以说出打开电灯，等待 MultiNet 识别成功返回后可以在 sample_length 内继续说出下一个命令词，比如 关闭电灯。  
 当配合唤醒使用时，如果用户在唤醒后需要连续识别多个命令词，推荐使用该模式。  
 
用户可以通过 `menuconfig -> ESP Speech Recognition -> speech commands recognition mode after wake up` 来对以上两种模式进行切换，默认为 SINGLE_RECOGNITION 模式。

注：CONTINUOUS_RECOGNITION 模式下对单个词的识别率略低于 SINGLE_RECOGNITION 模式下的单个词识别率。

wangxiankang, 2022-10-20:
0.8.0版本的sr关于识别模式的配置为：

choice CHINESE_SR_MN_MODEL_SEL
    prompt "Chinese Speech Commands Model"
    default SR_MN_CN_MULTINET4_5_SINGLE_RECOGNITION
    depends on USE_MULTINET
    help
        Select the Wake Word Engine to be used.

    config SR_MN_CN_NONE
        bool "None"

    config SR_MN_CN_MULTINET2_SINGLE_RECOGNITION
        bool "chinese single recognition (mn2_cn)"
        depends on IDF_TARGET_ESP32

    config SR_MN_CN_MULTINET4_5_SINGLE_RECOGNITION
        bool "chinese recognition (mn4_cn)"
        depends on IDF_TARGET_ESP32S3

    config SR_MN_CN_MULTINET4_5_SINGLE_RECOGNITION_QUANT8
        bool "chinese recognition (mn4q8_cn)"
        depends on IDF_TARGET_ESP32S3

    config SR_MN_CN_MULTINET5_RECOGNITION_QUANT8
        bool "english recognition (mn5q8_cn)"
        depends on IDF_TARGET_ESP32S3
endchoice

没找到关于CONTINUOUS_RECOGNITION相关配置，请大佬解答以下。谢谢

wangxiankang, 2022-10-20:
通过查看文档发现文档和当前0.8.0不符，文档修改者为 Wang Wang Wang.

文档路径：esp-adf/components/esp-sr/docs/speech_command_recognition.README_cn.md
发现，文档内是这样写的：

MultiNet 是为了在 ESP32 系列上离线实现多命令词识别而设计的轻量化模型，目前支持 200 个以内的自定义命令词识别。

> 支持中文和英文命令词识别（英文命令词识别需使用 ESP32S3）  
> 支持用户自定义命令词  
> 支持运行过程中 增加/删除/修改 命令词语  
> 最多支持 200 个命令词  
> 支持单次识别和连续识别两种模式  
> 轻量化，低资源消耗  
> 低延时，延时500ms内  
> 支持在线中英文模型切换（仅 ESP32S3）  
> 模型单独分区，支持用户应用 OTA

命令词识别支持两种基本模式：
> 单次识别  
> 连续识别
当命令词识别返回状态为 `ESP_MN_STATE_DETECTED` 时退出命令词识别，则为单次识别模式；  
当命令词识别返回状态为 `ESP_MN_STATE_TIMEOUT` 时退出命令词识别，则为连续识别模式；

feizi, 2022-10-21:
你好，如果你使用 ESP32_S3_KORVO2_V3 开发板， 建议直接使用master 分支，参考esp-skainet/examples,目前默认支持一次唤醒多次识别。
新的算法也会优先支持ESP32-S3。

wangxiankang, 2022-10-21:
> 你好，如果你在开发 ESP32_S3KORVO2_V3，建议直接使用新的分支，参考-skainet/examples，默认支持一次监控，并 支持 ESP32-S3。

感谢您的回复。我目前使用的是esp-adf master分支，我是基于[wwe demo](https://github.com/espressif/esp-adf/tree/master/examples/speech_recognition/wwe)进行开发调试的，我稍后看一下您说的skainet/examples。另外想问一下，在我当前的demo基础上如何修改可以支持到单次唤醒，多次识别那？

wangxiankang, 2022-10-22:
我还是想基于ADF去开发我的应用，参考的demo是 wwe、今天趁着周末看了一下adf下，recorder_sr.c 、audio_recorder.c 等相关代码，最后注释了一句代码实现了单次唤醒连续识别功能，具体方法是修改recorder_sr.c 文件的第193行附近：
if (mn_state == ESP_MN_STATE_DETECTED) {
            esp_mn_results_t *mn_result = multinet->get_results(recorder_sr->mn_handle);
            if (recorder_sr->mn_monitor) {
                recorder_sr->mn_monitor(mn_result->command_id[0], recorder_sr->mn_monitor_ctx); //cb -> thisself -> user 
            }
#if CONFIG_IDF_TARGET_ESP32
            recorder_sr_enable_wakenet_aec(recorder_sr);
#endif
            // detect_flag = 0;//停止检测 single detect model
        }

将上面detect_flag = 0; 注释掉即可。

closed 2022-10-22 by wangxiankang