espressif/esp-sr #57: undefined references when using aec API

boborjan2, 2023-01-15:
I get linker errors regarding undefined references to functions "esp_aec3_728_init", "esp_aec3_728_process", "esp_aec3_728_destroy" when using the aec API described in esp_aec.h (using aec_pro_create()/aec_create(), aec_process(), aec_destroy()).
This API was working a while ago. Are these still supported?

Thanks,
Viktor

feizi, 2023-01-17:
The API is deprecated. You can use AFE API to do AEC. If you do not need other algorithms in AFE, you can turn them off during initialization

boborjan2, 2023-01-17:
The problem with that approach is that all functions (aec,agc,ns,vad,etc) will be included in the binary (flash usage will be high). Thanks for the answer btw.

closed 2023-02-23 by feizi