espressif/esp-sr #58: 语音识别和blufi相冲突导致不断重启

casitelwithdean, 2023-01-31:
我使用的是esp32s3-eye开发板,单独开启语音识别正常，单独开启blufi也正常。当开启语音识别后再开启blufi，程序便会重启。log我放在了下面。我个人的愚见感觉是因为线程的问题吗？因为我看语音识别里面的feed_task和fetch_task都使用了xTaskCreatePinnedToCore分别绑定了0核和1核，而报错log里面也是有关线程的错误？
在此十分感谢。

 (11801) wifi_init: WiFi RX IRAM OP enabled
I (11802) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (11838) wifi:mode : sta (f4:12:fa:e1:93:d4)
I (11839) wifi:enable tsf
W (11851) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (11851) BT_INIT: BT controller compile version [421c279]
I (11856) BT_INIT: Bluetooth MAC: f4:12:fa:e1:93:d6

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4038c845  PS      : 0x00060633  A0      : 0x8038bd3d  A1      : 0x3fce8ec0
0x4038c845: uxListRemove at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/list.c:193

A2      : 0x3d80001c  A3      : 0x3fcaac44  A4      : 0x3fcaac44  A5      : 0x00060623
A6      : 0x00060620  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00014001
A10     : 0x3d8a2ff4  A11     : 0x00000804  A12     : 0x3c133fd8  A13     : 0x00000003
A14     : 0x02ca4b00  A15     : 0x00ffffff  SAR     : 0x0000000b  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00014005  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x4038c842:0x3fce8ec0 0x4038bd3a:0x3fce8ee0 0x4200cd50:0x3fce8f00 0x42010f26:0x3fce8f50 0x42011cbf:0x3fce8f80 0x42011d3e:0x3fce8fa0 0x4200d8e8:0x3fce8fc0 0x4200d984:0x3fce8fe0 0x4200b1b9:0x3fce9010 0x4200cb27:0x3fce9030 0x4038c8a5:0x3fce9060
0x4038c842: uxListRemove at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/list.c:192

0x4038bd3a: vTaskDelete at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/tasks.c:1340 (discriminator 4)

0x4200cd50: osi_thread_create at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/common/osi/thread.c:270

0x42010f26: hci_start_up at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/host/bluedroid/hci/hci_layer.c:109

0x42011cbf: bte_main_enable at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/host/bluedroid/main/bte_main.c:126

0x42011d3e: bte_main_boot_entry at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/host/bluedroid/main/bte_main.c:89

0x4200d8e8: btc_init_bluetooth at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/host/bluedroid/btc/core/btc_main.c:64

0x4200d984: btc_main_call_handler at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/host/bluedroid/btc/core/btc_main.c:111

0x4200b1b9: btc_thread_handler at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/common/btc/core/btc_task.c:200

0x4200cb27: osi_thread_run at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/bt/common/osi/thread.c:165

0x4038c8a5: vPortTaskWrapper at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 8060b2ea2eb06f3c

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b14
0x40375b14: esp_restart_noos at D:/esp-4.4.3/Espressif/frameworks/esp-idf-v4.4.3/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xb9c
load:0x403cc700,len:0x2e3c
entry 0x403c994c
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (24) boot: compile time 23:06:34
I (24) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (64) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (72) boot:  2 factory          factory app      00 00 00010000 00226000
I (79) boot:  3 ota_0            OTA app          00 10 00240000 000fa000
I (87) boot:  4 ota_1            OTA app          00 11 00340000 000fa000
I (94) boot:  5 model            Unknown data     01 82 0043a000 00320000
I (102) boot:  6 fr               Unknown data     01 06 0075a000 00020000
I (109) boot: End of partition table

feizi, 2023-02-01:
我问了同事， esp-sr和blufi可以一起运行。
你注意 task 退出一定要 vTaskDelete

casitelwithdean, 2023-02-01:
> 我问了同事， esp-sr和blufi可以一起运行。 你注意 task 退出一定要 vTaskDelete

谢谢回复。问题解决了，是通过使用较新版本的esp-sr解决的。

closed 2023-02-01 by casitelwithdean