espressif/esp-sr #60: AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

casitelwithdean, 2023-02-22:
Hi,我使用的是esp32s3wroom1 n8r8芯片，我只使用了1个mic，是和esp32s3-EYE一样的mic，代码也是直接使用esp-skainet，其中的ace_init我设置为了false：    afe_config.aec_init = false;。 然后再esp_afe_config.h里的mic通道设置如下：
    .pcm_config.total_ch_num = 2, \
    .pcm_config.mic_num = 1, \
    .pcm_config.ref_num = 1, \
 但是一直会显示AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!的warning。不过初始化之后仍然可以语音唤醒和识别指令。但是每隔五六次唤醒后又会有一次突然间弹出很多AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!，然后此次唤醒就识别不到指令了。

在此十分感谢！

I (1130) MODEL_LOADER:
Initializing models from SPIFFS, partition label: model

I (1334) MODEL_LOADER: Partition size: total: 2910596, used: 2085308

I (1511) I2S: DMA Malloc info, datalen=blocksize=640, dma_buf_count=6
I (1512) AFE_SR: afe interface for speech recognition

I (1514) AFE_SR: AFE version: SR_V220727

I (1518) AFE_SR: Initial auido front-end, total channel: 2, mic num: 1, ref num: 1

I (1527) AFE_SR: aec_init: 0, se_init: 1, vad_init: 1

I (1533) AFE_SR: wakenet_init: 1

model_name: wn9_xiaoaitongxue model_data: /srmodel/wn9_xiaoaitongxue/wn9_data
MC Quantized wakenet9: WakeNet9_v1h24_xiaoaitongxue_3_0.620_0.627, tigger:v3, mode:2, p:0, (Nov  3 2022 11:49:16)
I (2203) AFE_SR: wake num: 2, mode: 0, (Nov  3 2022 11:49:26)

multinet:mn4_cn
I (2221) MN: model_name: mn4_cn model_data: /srmodel/mn4_cn/mn4_cn_data

W (2674) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (2683) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2692) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2705) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2709) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2734) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2746) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2754) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2763) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2775) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2784) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2792) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2803) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2808) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2834) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2848) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2866) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2886) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2904) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2910) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2964) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (2983) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3003) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3015) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3021) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3080) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3100) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3120) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3130) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3136) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3200) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3218) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3237) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3251) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3256) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3324) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3343) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
W (3363) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
[... 127 lines cut ...]
I (4827) MN_COMMAND: Command ID20, phrase ID27: biao zhi deng hong se
W (4838) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

I (4841) MN_COMMAND: Command ID21, phrase ID28: biao zhi deng huang se
W (4851) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

I (4855) MN_COMMAND: Command ID22, phrase ID29: biao zhi deng lan se
W (4865) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

I (4868) MN_COMMAND: Command ID23, phrase ID30: biao zhi deng lv se
W (4879) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

I (4882) MN_COMMAND: Command ID24, phrase ID31: biao zhi deng qing se
W (4893) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

I (4896) MN_COMMAND: Command ID25, phrase ID32: biao zhi deng qing se
W (4907) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

I (4910) MN_COMMAND: ---------------------------------------------------------

W (4921) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (4935) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

------------detect start------------

BlueSkyB, 2023-02-23:
出现这个打印，代表库里的ringbuf满了。
ringbuf满了，就会导致音频丢帧，所以出现这个打印时，本次容易识别不了。

1. 首先启动时，就出现这个打印，是因为example代码里，识别multinet的初始化启动代码，是放在detect task中，导致afe_handle->fetch启动太慢，内部ringbuf溢出。可以参照英文识别的example代码，在启动detect task后，做些延迟，再启动feed task：

`   xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void*)afe_data, 5, NULL, 1);
    vTaskDelay(3000 / portTICK_PERIOD_MS);  // delay to load multinet model

    xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void*)afe_data, 5, NULL, 0);`

2. 同时，运行时还容易出现这个问题，可以调整 afe_config.afe_ringbuf_size 配置，增大内部ringbuf的大小。例如，在原有50基础上改为150，就是放大了 3 倍。

3. 如果是自有程序，非demo，要保证detect task中不能有过多耗时动作，例如太多的逻辑操作、写flash操作、延时操作等。应保证afe_handle->fetch 取数据的及时性。

casitelwithdean, 2023-02-23:
> 出现这个打印，代表库里的ringbuf满了。 ringbuf满了，就会导致音频丢帧，所以出现这个打印时，本次容易识别不了。
>
> 1. 首先启动时，就出现这个打印，是因为example代码里，识别multinet的初始化启动代码，是放在detect task中，导致afe_handle->fetch启动太慢，内部ringbuf溢出。可以参照英文识别的example代码，在启动detect [...]

谢谢您的回复，非常感谢！

closed 2023-03-06 by casitelwithdean