espressif/esp-sr #61: srmodel_mmap_init crash on esp32s3 with esp-idf 4.4.4

waruqi, 2023-03-27:
I tested https://github.com/espressif/esp-skainet/tree/master/examples/wake_word_detection on my esp32s3

But it will crash in esp-sr.

```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759e8
0x403759e8: esp_restart_noos at /Users/ruki/projects/personal/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbcc
load:0x403cc700,len:0x2fac
entry 0x403c9954
I (29) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (29) boot: compile time 16:44:26
I (29) boot: chip revision: v0.1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=28704h (165636) map
I (140) esp_image: segment 1: paddr=0003872c vaddr=3fc94760 size=04eb0h ( 20144) load
I (145) esp_image: segment 2: paddr=0003d5e4 vaddr=40374000 size=02a34h ( 10804) load
I (149) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=74348h (475976) map
I (240) esp_image: segment 4: paddr=000b4370 vaddr=40376a34 size=0dd24h ( 56612) load
I (260) boot: Loaded app from partition at offset 0x10000
I (260) boot: Disabling RNG early entropy source...
I (271) cpu_start: Pro cpu up.
I (271) cpu_start: Starting app cpu, entry point is 0x40375434
0x40375434: call_start_cpu1 at /Users/ruki/projects/personal/esp-idf/components/esp_system/port/cpu_start.c:148

I (250) cpu_start: App cpu up.
I (285) cpu_start: Pro cpu start user code
I (286) cpu_start: cpu freq: 160000000
I (286) cpu_start: Application information:
I (288) cpu_start: Project name:     OttoBot
I (293) cpu_start: App version:      07714a2-dirty
I (299) cpu_start: Compile time:     Mar 27 2023 22:02:33
I (305) cpu_start: ELF file SHA256:  cba559b5716d2cd6...
I (311) cpu_start: ESP-IDF:          v4.4.4
I (316) heap_init: Initializing. RAM available for dynamic allocation:
I (323) heap_init: At 3FC9B6B8 len 0004E058 (312 KiB): D/IRAM
I (329) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (336) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (342) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (349) spi_flash: detected chip: generic
I (353) spi_flash: flash io: dio
W (357) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (371) sleep: Configure to isolate all GPIO pins in sleep state
I (377) sleep: Enable automatic switching of GPIO sleep configuration
I (384) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (405) wakeup: init wakeup
I (405) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (415) I2S: I2S1, MCLK output by GPIO20
I (415) I2S: DMA Malloc info, datalen=blocksize=320, dma_buf_count=6
I (425) I2S: I2S0, MCLK output by GPIO42
W (425) board: Codec not detected
I (435) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (445) wakeup: esp_srmodel_init
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420073a9  PS      : 0x00060730  A0      : 0x8200750a  A1      : 0x3fcf3a30
0x420073a9: srmodel_mmap_init at /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/src/model_path.c:248

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fc99d4c  A5      : 0x3fcf3a80
A6      : 0x3fcf3a60  A7      : 0x3fcea088  A8      : 0x8200739a  A9      : 0x3fcf3a10
A10     : 0x3fcea088  A11     : 0x000000ff  A12     : 0x3c083664  A13     : 0x00000000
A14     : 0x3fcf3a40  A15     : 0x3fcea090  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x420073a6:0x3fcf3a30 0x42007507:0x3fcf3a80 0x4200727a:0x3fcf3aa0 0x420070ff:0x3fcf3b30 0x420070f7:0x3fcf3b50 0x42074178:0x3fcf3b70 0x4037ea61:0x3fcf3b90
0x420073a6: srmodel_mmap_init at /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/src/model_path.c:248

0x42007507: esp_srmodel_init at /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/src/model_path.c:460

0x4200727a: ot_speech_wakeup_test at /Users/ruki/projects/personal/hellokiki/ottobot/main/tests/speech_wakeup.c:108 (discriminator 13)

0x420070ff: ot_tests_main at /Users/ruki/projects/personal/hellokiki/ottobot/main/tests_main.c:32

0x420070f7: app_main at /Users/ruki/projects/personal/hellokiki/ottobot/main/main.c:38

0x42074178: main_task at /Users/ruki/projects/personal/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

0x4037ea61: vPortTaskWrapper at /Users/ruki/projects/personal/esp-idf/components/freertos/port/xtensa/port.c:142

ELF file SHA256: cba559b5716d2cd6

```

waruqi, 2023-03-27:
I used this commit https://github.com/espressif/esp-sr/commit/a4f33325610c516482aeb168e508e6e074e2f251

feizi, 2023-03-31:
Hi @waruqi , Please update the lates esp-sr.
Now it works fine on both idf-5v.0 and idf-v4.4

waruqi, 2023-03-31:
> Hi @waruqi , Please update the lates esp-sr. Now it works fine on both idf-5v.0 and idf-v4.4

Ok thanks, I will try it later.

waruqi, 2023-03-31:
It works, thanks.

waruqi, 2023-04-01:
I tried it, it will compile errors for esp-idf 4.4.4.

```
omponents/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/ruki/projects/personal/hellokiki/ottobot/build/bootloader/bootloader.bin
Bootloader binary size 0x5230 bytes. 0x2dd0 bytes (36%) free.
[8/10] Linking CXX executable OttoBot.elf
FAILED: OttoBot.elf
: && /Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls  CMakeFiles/OttoBot.elf.dir/project_elf_src_esp32s3.c.obj -o Ot
toBot.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf
/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub
/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  e
sp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wif
i/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/
libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a
 esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc
_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp
_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a
 esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_
levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/l
ibopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/usb/libusb.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/espressif__esp-dsp/libesp
ressif__esp-dsp.a  esp-idf/esp-sr/libesp-sr.a  esp-idf/led_strip/libled_strip.a  esp-idf/hardware_driver/libhardware_driver.a  esp-idf/player/libplayer.a  esp-idf/main/libmain.a  -Wl,--cref  -Wl,--M
ap="/Users/ruki/projects/personal/hellokiki/ottobot/build/OttoBot.map"  -Wl,--gc-sections  -fno-rtti  -fno-lto  esp-idf/unity/libunity.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprot
obuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/esp-sr/libesp-sr.a  /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libhufzip.a  /Users/ruki/projects/personal/hellokiki/ottob
ot/components/esp-sr/lib/esp32s3/libesp_audio_front_end.a  /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_processor.a  /Users/ruki/projects/personal/hello
kiki/ottobot/components/esp-sr/lib/esp32s3/libmultinet.a  /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libwakenet.a  esp-idf/esp-sr/libesp-sr.a  /Users/ruki/projects
/personal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libhufzip.a  /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_front_end.a  /Users/ruki/projects/pe
rsonal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_processor.a  /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libmultinet.a  /Users/ruki/projects/per
sonal/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libwakenet.a  esp-idf/json/libjson.a  esp-idf/espressif__esp-dsp/libespressif__esp-dsp.a  -L /Users/ruki/projects/personal/hellokiki/ottobot/com
ponents/esp-sr/lib/esp32s3  -L /Users/ruki/projects/personal/hellokiki/ottobot/components/esp-sr/esp-tts/esp_tts_chinese/esp32s3  -Wl,--start-group  -ldl_lib  -lfst  -lc_speech_features  esp-idf/esp
ressif__esp-dsp/libespressif__esp-dsp.a  -lesp_tts_chinese  -lvoice_set_xiaole  -Wl,--end-group  esp-idf/led_strip/libled_strip.a  esp-idf/player/libplayer.a  esp-idf/hardware_driver/libhardware_dri
ver.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/spiffs/libspiffs.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a
 esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libsp
[... cut ...]
l/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj): in function `rtc::FatalMessage::FatalMessage(char const*, int)':
/home/sunxiangyu/workspace/esp_sr_lib/build/../components/esp_audio_processor/rtc_base/checks.cc:69: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allo
cator<char> >::basic_ostringstream()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

closed 2024-04-10 by feizi