espressif/esp-sr #62: 运行太多任务，导致语音识别率急剧降低

casitelwithdean, 2023-04-04:
Hi，我正在使用esp32s3wroomn8r8，和esp32s3 eye同样的麦克风，语音模型使用的是小爱同学 mn5 q8cn
我自定义了大概40个指令，如果只运行sr的task，那么识别率是非常高的。
但是现在我加入了blufi，wifi，led-strip，touchpad，sigmadelta以及4路adc1。加上之后识别率就会飘忽不定，有时还可以，有时就怎么都识别不出来。
Blufi绑定的core 0，  wifi绑定的core 1，detect_task绑定的core1   ，feed_task绑定的core  0 
个人感觉是因为语音识别的时候，要同时执行太多任务，所以没办法一直处理语音数据，所以才导致识别率下降
请问有什么办法可以改善吗？还是说并不是多任务引起的，而是由其他因素引起的呢？
在此十分感谢！

feizi, 2023-04-04:
Hi @casitelwithdean ,运行多任务是可能导致性能下降，主要的问题就是 **掉帧**。当 wakenet,multinet没有足够的CPU资源运行时，就是随机丢弃ringbuffer中的部分音频。  
关于这个问题的调试，你可以检查log中是否有类似 “ERROR! rb_in slow” 或者 “ERROR! rb_out slow!!!”

casitelwithdean, 2023-04-04:
> Hi @casitelwithdean ,运行多任务是可能导致性能下降，主要的问题就是 **掉帧**。当 wakenet,multinet没有足够的CPU资源运行时，就是随机丢弃ringbuffer中的部分音频。 关于这个问题的调试，你可以检查log中是否有类似 “ERROR! rb_in slow” 或者 “ERROR! rb_out slow!!!”

Hi，多谢回复。
起初是存在error rb slow的log的，后来通过[https://github.com/espressif/esp-sr/issues/60](url)   这个帖子，增大了ring buffer size到200，解决了error rb slow的问题。
虽然解决了rb slow的问题，但仍然识别率不理想。

feizi, 2023-04-04:
更新到最新的esp-sr, 你可以试试 mn6_cn性能如何

closed 2023-04-24 by casitelwithdean