espressif/esp-sr #63: 最新1.3版本的是不是打包a文件的时候指定了/home/sunxiangyu/workspace/esp_sr_lib的工作路径

JedTK, 2023-04-06:
[image]
这让原来可以运行的项目无法进行编译

feizi, 2023-04-06:
应该不是这个问题，1.3版本更新对 idf v5.0的支持，保留了对idf v4.4的兼容，但是去除了对idf v3.3的支持
你可以试试`idf.py fullclean` 然后在 `idf.py build`

JedTK, 2023-04-06:
我已经按照你说的：idf.py fullclean 然后在 idf.py build
但是还是一样的错误，依然引入的是 /home/sunxiangyu/workspace/esp_sr_lib/

JedTK, 2023-04-06:
我这边引入了c++的支持依然还是无法编译通过，然后尝试将之前版本/esp-sr/lib/下的所有a文件都复制到新的版本的路径上是没问题的，应该还是编译a文件的时候，你们固定了c++支持的inclu路径

feizi, 2023-04-06:
这个问题是最新的代码使用更新版本的GCC编译，导致在旧的GCC上出现不兼容。具体可参考[这篇博客](https://blog.csdn.net/github_28260175/article/details/105922130)

为什么会出现两个不同版本的GCC?
因为新的代码是在idf v5.0下编译，idf v5.0更新了GCC的版本，和idf v4.4或更老的idf GCC版本不同了。

closed 2023-04-07 by feizi

urpp, 2024-08-19:
这个sunxiangyu..我也遇到了..请问怎么解决?
[image]

feizi, 2024-08-20:
这个路径不是问题， 问题是` undefined reference to 'dsps_bit_rev4r_fc32'`, 
你在[CMakeLists.txt](https://github.com/espressif/esp-sr/blob/ffcd69533f238e128300b64be4a9e886022c39ef/CMakeLists.txt#L160)这个地方添加
add_prebuilt_library(nsnet "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32p4/libc_speech_feature.a" PRIV_REQUIRES ${COMPONENT_NAME})