espressif/esp-sr #64: IDF 4.4 incompatibility for esp32s2 and esp32c3 variants

bmichini, 2023-04-06:
In `CMakeLists.txt` the IDF 4.4 compatibility is fixed, but only for [esp32](https://github.com/espressif/esp-sr/blob/f136e7f11c6c18ffaf68ab7b803f46319bd3076a/CMakeLists.txt#L18) and [esp32s3](https://github.com/espressif/esp-sr/blob/f136e7f11c6c18ffaf68ab7b803f46319bd3076a/CMakeLists.txt#L63) variants.

The [esp32s2](https://github.com/espressif/esp-sr/blob/f136e7f11c6c18ffaf68ab7b803f46319bd3076a/CMakeLists.txt#L123) and [esp32c3](https://github.com/espressif/esp-sr/blob/f136e7f11c6c18ffaf68ab7b803f46319bd3076a/CMakeLists.txt#L134) builds still require `esp_partition` by default, which causes build errors in IDF 4.4.

feizi, 2023-04-07:
Thanks for the reminder. I will fixed it.

feizi, 2023-04-23:
The bug has been fixed

closed 2023-04-23 by feizi