espressif/esp-sr #65: mn6_cn  how to add or modify the command when esp32s3 is running?

casitelwithdean, 2023-04-22:
Hi,i am using esp32s3-wroom-n8r8,and I want modify some commands when esp32s3 is running,and I tried the following codes,but it doesn't work.

     char *cn_commands = "guan bi kong tiao";
     esp_mn_commands_alloc();
     esp_mn_commands_add(0,cn_commands );

So how to modify the command  in mn6_cn ?
Thanks a lot

feizi, 2023-04-23:
Currently mn6_cn does not support modifying command words on the fly.

feizi, 2023-05-25:
Now , MultiNet can add/modify/remove commands in the code.
Please refer to [API](https://github.com/espressif/esp-sr/blob/master/src/esp_mn_speech_commands.c) and [document](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/speech_command_recognition/README.html#customize-speech-commands-via-api-calls)

closed 2023-05-25 by feizi

casitelwithdean, 2023-05-25:
> Hi @casitelwithdean , Now multinet can add/modify/remove/search commands in the code.
> 
> All speech commands are save in a singly linked list. It is easy to add one speech command into linked list and remove one speech command from linked list. The code was implemented in [esp_mn_speech_commands.c](https://github.com/espressif/esp-sr/blob/master/src/esp_mn_speech_commands.c). You also can modify the code by yourself. The document of API is [here](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/speech_command_recognition/README.html#customize-speech-commands-via-api-calls).

Thanks very much !!!

wangruiran, 2025-04-27:
你好，我想定制新的可以替换的唤醒词为“你好，小纤”运用到ESP32-S3-BOX-3B的语言系统中，用于后续中和我们自己设置的模型进行沟通。我的邮件为1602518893@qq.com或者wangruiran43@gmail.com进行联系。感谢您的帮助，我们可以详细交流设计过程和相应的费用问题