espressif/esp-sr #69: using multinet6-cn with esp-box panics

IamFive, 2023-05-16:
panic occurs when run to `multinet->create(mn_name, 5760);`, same code works when loading multinet6-en models.

```
Build fst from commands.
Quantized MultiNet6:rnnt_ctc_1.0, name:mn6_cn, (May  5 2023 20:32:52)
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40386816  PS      : 0x00060533  A0      : 0x80386ec5  A1      : 0x3fca3930
0x40386816: insert_free_block at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/heap_tlsf.c:237
 (inlined by) block_insert at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/heap_tlsf.c:263
 (inlined by) tlsf_free at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/heap_tlsf.c:968

A2      : 0x3fc9fd9c  A3      : 0xbfc9fd9c  A4      : 0x3fc9efcc  A5      : 0x00060523
A6      : 0x00060520  A7      : 0x00000001  A8      : 0x3fca9028  A9      : 0x00000000
A10     : 0x0000000a  A11     : 0x3fca9030  A12     : 0x3fc9fdf4  A13     : 0x00000005
A14     : 0x02ca048c  A15     : 0x00ffffff  SAR     : 0x00000001  EXCCAUSE: 0x0000001d
EXCVADDR: 0xbfc9fda8  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd

Backtrace: 0x40386813:0x3fca3930 0x40386ec2:0x3fca3950 0x4037a046:0x3fca3970 0x40387329:0x3fca3990 0x42021a03:0x3fca39b0 0x420219d5:0x3fca39d0 0x420219d5:0x3fca39f0 0x42017c08:0x3fca3a10 0x42017caf:0x3fca3d8
0 0x42006918:0x3fca3da0 0x42006e01:0x3fca3dd0 0x4200634d:0x3fca3e20 0x4206bcc0:0x3fca3e50 0x40383425:0x3fca3e70
0x40386813: insert_free_block at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/heap_tlsf.c:236
 (inlined by) block_insert at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/heap_tlsf.c:263
 (inlined by) tlsf_free at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/heap_tlsf.c:968

0x40386ec2: multi_heap_free_impl at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/multi_heap.c:225

0x4037a046: heap_caps_free at D:/espressif/frameworks/esp-idf-v4.4.4/components/heap/heap_caps.c:361

0x40387329: free at D:/espressif/frameworks/esp-idf-v4.4.4/components/newlib/heap.c:39

0x42021a03: cJSON_Delete at D:/espressif/frameworks/esp-idf-v4.4.4/components/json/cJSON/cJSON.c:271

0x420219d5: cJSON_Delete at D:/espressif/frameworks/esp-idf-v4.4.4/components/json/cJSON/cJSON.c:261

0x420219d5: cJSON_Delete at D:/espressif/frameworks/esp-idf-v4.4.4/components/json/cJSON/cJSON.c:261

0x42017c08: model_create at /home/sunxiangyu/workspace/esp_sr_lib/build/../components/multinet/multinet6_quantized.c:980

0x42017caf: model_init at /home/sunxiangyu/workspace/esp_sr_lib/build/../components/multinet/multinet6_quantized.c:1831

```

feizi, 2023-05-25:
I will check it.

feizi, 2023-05-31:
I reproduced the same log. This is because there are more than 300 default speech commands for Chinese MultiNet6, but less than 100 for English. This results in the need for larger **stack** memory when initializing the Chinese model. 
You can solve this problem by increasing the stack memory or deleting some default speech commands for Chinese.

closed 2023-06-01 by IamFive