espressif/esp-sr #70: does esp-sr supports recognize speech and output text

IamFive, 2023-05-18:
esp-sr now supports speech to offline command(custom command) recognization, but i did not find any api that supports speech to text.
like we hope if no offline command matches speech, we can get the text command, and start a customer handle.

IamFive, 2023-05-18:
@feizi 似乎esp-sr的issue无法同步到你们的jira上，请帮忙看下这个问题。

feizi, 2023-05-25:
> esp-sr now supports speech to offline command(custom command) recognization, but i did not find any api that supports speech to text. like we hope if no offline command matches speech, we can get the text command, and start a customer handle.

Yes, we do not have API to get the text of speech. We've done some fuzzy matching in beam searching.  
Of course, if you think it is necessary for you, we can add an API to output the results of the greed searching when no commands are recognized

IamFive, 2023-05-25:
> Yes, we do not have API to get the text of speech. We've done some fuzzy matching in beam searching.
> Of course, if you think it is necessary for you, we can add an API to output the results of the greed searching when no commands are recognized

it will be great if we can get the speech text. And I think it is very necessary for speech fallback handle if no command can be detect. please update the esp-sr.

feizi, 2023-06-02:
When mn_state is `ESP_MN_STATE_TIMEOUT` or `ESP_MN_STATE_DETECTED`, the text of speech can be obtained as following method:

```
esp_mn_state_t mn_state = multinet->detect(model_data, res->data);
if (mn_state == ESP_MN_STATE_TIMEOUT ||  mn_state == ESP_MN_STATE_DETECTED) 
{
    esp_mn_results_t *mn_result = multinet->get_results(model_data);
    printf("timeout, text:%s\n", mn_result->string);
}

```

IamFive, 2023-06-05:
> 

No matter the mn command is detected or not, when log mn_result, it outputs something like: 
```
EmE0�ER�E�UE��E'E��E��ES\EB�E�'Ei�E��ESTEt�E
                                            E|E��E�>E�E�E�^Ey�E�E�yE`�E_4E�E��E�HE��EJ�E~XEA�E�
                                                                                               EodE޼E�EslE��EVE�pE��EE(qE��E(EnE��E� Eyg Eٹ E�
                                                                                                                                              !Ez
]!E��!E��!E,P"EZ�"E,�"E�?#EĎ#E6�?

```

I guess there are some mem error in code.

feizi, 2023-06-05:
I test this function by esp-skainet. And it works fine.
Please refer to [en_speech_commands_recognition](https://github.com/espressif/esp-skainet/blob/master/examples/en_speech_commands_recognition/main/main.c#LL122C1-L127C26)

IamFive, 2023-06-05:
> I test this function by esp-skainet. And it works fine. Please refer to [en_speech_commands_recognition](https://github.com/espressif/esp-skainet/blob/master/examples/en_speech_commands_recognition/main/main.c#LL122C1-L127C26)

ok, will test it again. thanks.

IamFive, 2023-06-05:
although, i try flash the [https://github.com/espressif/esp-skainet/blob/master/examples/en_speech_commands_recognition](example ) you provided.

it results:

```
I (291) esp_image: segment 4: paddr=0014a7c4 vaddr=403817I (712) esp_psram: SPI SRAM memory test OK
I (721) cpu_start: Pro cpu start user code
I (721) cpu_start: cpu freq: 240000000 Hz
I (721) cpu_start: Application information:
I (724) cpu_start: Project name:     speech_commands_recognition
I (731) cpu_start: App version:      v0.3.0-195-ga48307f
I (737) cpu_start: Compile time:     Jun  5 2023 15:36:26
I (743) cpu_start: ELF file SHA256:  083b949fbe0f7a2b...
I (749) cpu_start: ESP-IDF:          v5.0.2-dirty
I (754) cpu_start: Min chip rev:     v0.0
I (759) cpu_start: Max chip rev:     v0.99
I (764) cpu_start: Chip rev:         v0.1
I (768) heap_init: Initializing. RAM available for dynamic allocation:
I (775) heap_init: At 3FCA0940 len 00048DD0 (291 KiB): DRAM
I (782) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (788) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (795) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (803) spi_flash: detected chip: gd
I (806) spi_flash: flash io: qio
W (811) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (821) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (842) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (842) MODEL_LOADER: The storage free size is 6912 KB
I (852) MODEL_LOADER: The partition size is 5168 KB
I (862) MODEL_LOADER: Successfully map model partition
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect............
```

kristiankielhofner, 2023-06-05:
For whatever it is worth we are using this successfully with multinet 6, it's likely your issue is related to multinet 5. We've also seen much better results with multinet 6 and from looking at the commit history on esp-sr multinet 6 is clearly the future. I wouldn't expect multinet 5 to support much/any new functionality.

closed 2023-06-06 by IamFive