espressif/esp-sr #71: Request for Adding Support for Other Sound Detection (e.g., Crying)

xandyxor, 2023-05-30:
Hello! I am very interested in the ESP-SR project and I am impressed with its sound detection capabilities. According to the project documentation, it already supports wake word detection and voice command recognition.

However, I would like to know if it would be possible to add support for other sound detection in the framework, specifically for crying sound. Crying sound detection can be very useful in applications such as baby monitoring and home security.

Integrating a crying sound detection algorithm into the ESP-SR framework would make it more comprehensive and practical. This feature would enable developers to build a wider range of applications and provide users with more options.

Thank you and your team for your contributions to the open-source community. I look forward to your response and feedback.

feizi, 2023-05-30:
The wake wake model is fully capable of sound detection. The main difficulty is to collect crying samples.  
We will try to collect crying sound samples. When we have enough samples, we will release this model.

xandyxor, 2023-05-30:
抱歉我用中文可能比較能表達較明確
請問哪裡可以看到如何 自行訓練非“語言（中文 or 英文）”的聲音 如哭聲等等

I can express myself more clearly in Chinese. Where can I find information on training non-verbal(eng or chinese) sounds, such as crying sounds, by myself?

feizi, 2023-05-30:
当前我们还没有开源 训练部分的代码
就我知道的，在espressif有一个团队准备去开源一个sound detection training and deployment projec.

feizi, 2024-04-10:
@xandyxor ， please refer to https://github.com/espressif/esp-sr/issues/88 to customize your own wake words

closed 2024-04-10 by feizi