espressif/esp-sr #74: Human voice activity detection

tareqAmen, 2023-07-29:
can any one help me to build esp32 code make human voice activity detection using esp-sr framework?
In my project, I want to turn on my device when there is only a human voice, so that it is not affected by any other sound, such as the sound of the wind or any other sounds in the environment.

sun-xiangyu, 2023-08-08:
You can get vad flag by runing `afe_fetch_result_t* res = afe_handle->fetch(afe_data);` Please refer to [here](https://github.com/espressif/esp-skainet/blob/a9429c33cf4a2e9f8b610eb0abb676dede23f59f/examples/cn_speech_commands_recognition/main/main.c#L90C9-L90C27)

Please refer to [here ](https://github.com/espressif/esp-sr/blob/455314a90cac59d4f50253cf719659f0b9f5d778/include/esp32s3/esp_afe_sr_iface.h#L21C1-L35C22) to get details of afe_fetch_result_t

jayavanth, 2023-08-09:
@sun-xiangyu Not the OP but that was very helpful

Do I need the Wakenet model if I just want to do VAD? I noticed that to do VAD, I had to set `.wakenet_init = true` otherwise it won't work

sun-xiangyu, 2023-08-09:
You do not need to enable WakeNet. But you need to set `.se_init = false`.

jayavanth, 2023-08-09:
@sun-xiangyu I tried that but it doesn't seem to work. Here's what I have

```
    afe_config_t afe_config = AFE_CONFIG_DEFAULT();
    afe_config.memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;
    afe_config.wakenet_init = false;
    afe_config.se_init= false;
    afe_config.voice_communication_init = false;
    afe_config.vad_mode =  VAD_MODE_4;
    afe_config.aec_init = false;
    afe_config.pcm_config.total_ch_num = 2;
    afe_config.pcm_config.mic_num = 1;
    afe_config.pcm_config.ref_num = 1;

```

here's AFE_CONFIG_DEFAULT

```
#define AFE_CONFIG_DEFAULT() { \
    // Configures whether or not to enable AEC
    .aec_init = true, \
    // Configures whether or not to enable BSS/NS
    .se_init = true, \
    // Configures whether or not to enable VAD (only for speech recognition)
    .vad_init = true, \
    // Configures whether or not to enable WakeNet
    .wakenet_init = true, \
    // Configures whether or not to enable voice communication (cannot be enabled when wakenet_init is also enabled)
    .voice_communication_init = false, \
    // Configures whether or not to enable AGC for voice communication
    .voice_communication_agc_init = false, \
    // Configures the AGC gain (unit: dB)
    .voice_communication_agc_gain = 15, \
    // Configures the VAD mode (the larger the number is, the more aggressive VAD is)
    .vad_mode = VAD_MODE_3, \
    // Configures the wake model. See details below.
    .wakenet_model_name = NULL, \
    // Configures the wake mode. (corresponding to wakeup channels. This should be configured based on the number of mic channels)
    .wakenet_mode = DET_MODE_2CH_90, \
    // Configures AFE mode (SR_MODE_LOW_COST or SR_MODE_HIGH_PERF)
    .afe_mode = SR_MODE_LOW_COST, \
    // Configures the internal BSS/NS/MISO algorithm of AFE will be running on which CPU core
    .afe_perferred_core = 0, \
    // Configures the priority of BSS/NS/MISO algorithm tasks
    .afe_perferred_priority = 5, \
    // Configures the internal ringbuf size
    .afe_ringbuf_size = 50, \
    // Configures the memory allocation mode. See details below.
    .memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM, \
    // Configures the linear audio amplification level. See details below.
    .agc_mode = AFE_MN_PEAK_AGC_MODE_2, \
    // Configures the total number of audio channels
    .pcm_config.total_ch_num = 3, \
    // Configures the number of microphone channels
    .pcm_config.mic_num = 2, \
    // Configures the number of reference channels
    .pcm_config.ref_num = 1, \
}
```

BlueSkyB, 2023-08-09:
@jayavanth  It's a bug that vad doesn't work while disable WakeNet. We will fix it.
I noticed that your channel configuration is as follows:
`    afe_config.pcm_config.total_ch_num = 2;
    afe_config.pcm_config.mic_num = 1;
    afe_config.pcm_config.ref_num = 1;`
In the case of `mic_num = 1`, there is no need to set `.se_ Init=false`. On the contrary, set `.se_ Init=true`, the effect may be better because it can eliminate some steady-state noise.

The VAD of esp-sr does not distinguish between human voice and environmental noise, it is only related to the audio energy. Therefore, excessive environmental noise may also trigger VAD.

jayavanth, 2023-08-09:
Thanks @BlueSkyB! Will be on the lookout for the fix. Great! Will try the `true` setting

feizi, 2023-08-15:
@jayavanth, this bug has beed fixed.  please try again.

jayavanth, 2023-08-16:
Thanks @feizi will try it and will report back soon

closed 2023-08-21 by feizi