espressif/esp-sr #76: Having a hard time linking esp-sr to my project

jayavanth, 2023-08-10:
I have added esp-sr to my `main/CMakeLists.txt`
```
set(requires
 esp-sr
...
)
```

and I have also added it as EXTRA_COMPONENT_DIRS in my CMakeLists.txt
```
set(EXTRA_COMPONENT_DIRS PATH/esp-sr)
```

But I'm still getting this `ld` error:

```
/Users/jay/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(main.cc.obj):(.literal._Z13recv_from_x86Pv+0x8c): undefined reference to `_Z16esp_srmodel_initPKc'
/Users/jay/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(main.cc.obj):(.literal._Z13recv_from_x86Pv+0x90): undefined reference to `_Z18esp_srmodel_filterP14srmodel_list_tPKcS2_'
/Users/jay/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(main.cc.obj): in function `_Z13recv_from_x86Pv':
zmain.cc:487: undefined reference to `_Z16esp_srmodel_initPKc'
/Users/jay/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: PROJECTPATH/main.cc:488: undefined reference to `_Z18esp_srmodel_filterP14srmodel_list_tPKcS2_'
collect2: error: ld returned 1 exit status
```

I also can't seem to find those symbols when I search in `esp-sr/lib/esp32s3/*.a`. Can someone please tell me what I'm missing? Thank you!

igrr, 2023-08-10:
This issue is probably due to the fact that your main source file is a C++ file, but the header files in this repository are missing C++ guards. This should probably be fixed. In the meantime it should be possible to work around the issue by surrounding every `#include` statement in your main.cc file related to esp-sr with
```
extern "C" {
#include "..."
}
```

jayavanth, 2023-08-10:
@igrr OMG thank you that worked!!

igrr, 2023-08-10:
Thanks for confirming! Let's keep this open until this is fixed in the repo.

feizi, 2023-08-11:
I will check and fix this issue.

feizi, 2023-10-11:
The issue has been fixed.

closed 2023-10-11 by feizi