espressif/esp-sr #83: undefined reference to `__wrap_longjmp'

hornej, 2023-11-19:
I am trying to use espressif__esp-dsp and espressif__esp-sr in my project and I am getting **undefined reference to `__wrap_longjmp'** when trying to build. I looked at this [issue](https://github.com/espressif/esp-idf/issues/10496) but couldn't understand how I could apply the fix to my project. Is this an issue with esp-sr or a linking problem in my cmake files or something? 

This is my first time using ESP32 and ESP-IDF and have looked around but haven't been able to find anything to figure this out. I am using IDF v5.1.1

The only file I have modified is main.c so far

```
#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_wn_iface.h"
#include "esp_wn_models.h"
#include "esp_afe_sr_iface.h"
#include "esp_afe_sr_models.h"
#include "esp_mn_iface.h"
#include "esp_mn_models.h"
#include "model_path.h"
#include "esp_process_sdkconfig.h"
#include "driver/i2s_pdm.h"
#include "esp_err.h"

#define CONFIG_EXAMPLE_I2S_CLK_GPIO  4
#define CONFIG_EXAMPLE_I2S_DATA_GPIO 15
#define CONFIG_EXAMPLE_SAMPLE_RATE   16000
#define ADC_I2S_CHANNEL 4  

esp_afe_sr_data_t *afe_data = NULL;

i2s_chan_handle_t rx_handle = NULL;

int detect_flag = 0;
static esp_afe_sr_iface_t *afe_handle = NULL;
static volatile int task_flag = 0;
srmodel_list_t *models = NULL;
static int play_voice = -2;

void init_microphone(void)
{
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);
    ESP_ERROR_CHECK(i2s_new_channel(&chan_cfg, NULL, &rx_handle));

    i2s_pdm_rx_config_t pdm_rx_cfg = {
        .clk_cfg = I2S_PDM_RX_CLK_DEFAULT_CONFIG(CONFIG_EXAMPLE_SAMPLE_RATE),
        /* The default mono slot is the left slot (whose 'select pin' of the PDM microphone is pulled down) */
        .slot_cfg = I2S_PDM_RX_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO),
        .gpio_cfg = {
            .clk = CONFIG_EXAMPLE_I2S_CLK_GPIO,
            .din = CONFIG_EXAMPLE_I2S_DATA_GPIO,
            .invert_flags = {
                .clk_inv = false,
            },
        },
    };
    ESP_ERROR_CHECK(i2s_channel_init_pdm_rx_mode(rx_handle, &pdm_rx_cfg));
    ESP_ERROR_CHECK(i2s_channel_enable(rx_handle));
}

void init_esp_sr(void)
{
    esp_afe_sr_iface_t *afe_handle = &ESP_AFE_SR_HANDLE;
    afe_config_t afe_config = AFE_CONFIG_DEFAULT();
    afe_config.wakenet_model_name = esp_srmodel_filter(models, ESP_WN_PREFIX, NULL);
    afe_data = afe_handle->create_from_config(&afe_config);
}

esp_err_t bsp_get_feed_data(bool is_get_raw_channel, int16_t *buffer, int buffer_len)
{
    esp_err_t ret = ESP_OK;
    size_t bytes_read;

    // Read the audio data from the I2S interface
    ret = i2s_channel_read(rx_handle, (void *)buffer, buffer_len, &bytes_read, portMAX_DELAY);
    if (ret != ESP_OK) {
        printf("Error reading from I2S interface: %s\n", esp_err_to_name(ret));
        return ret;
    }

    // If not getting raw channel data, process the buffer here as needed

    return ret;
}

int bsp_get_feed_channel(void)
{
    return ADC_I2S_CHANNEL;
}

void feed_Task(void *arg)
{
    afe_data = arg;
    int audio_chunksize = afe_handle->get_feed_chunksize(afe_data);
    int nch = afe_handle->get_channel_num(afe_data);
    int feed_channel = bsp_get_feed_channel();
    assert(nch <= feed_channel);
    int16_t *i2s_buff = malloc(audio_chunksize * sizeof(int16_t) * feed_channel);
    assert(i2s_buff);

    while (task_flag) {
        bsp_get_feed_data(false, i2s_buff, audio_chunksize * sizeof(int16_t) * feed_channel);

        afe_handle->feed(afe_data, i2s_buff);
    }
    if (i2s_buff) {
        free(i2s_buff);
        i2s_buff = NULL;
    }
    vTaskDelete(NULL);
}

void detect_Task(void *arg)
{
    esp_afe_sr_data_t *afe_data = arg;
    int afe_chunksize = afe_handle->get_fetch_chunksize(afe_data);
    char *mn_name = esp_srmodel_filter(models, ESP_MN_PREFIX, ESP_MN_ENGLISH);
    printf("multinet:%s\n", mn_name);
    esp_mn_iface_t *multinet = esp_mn_handle_from_name(mn_name);
    model_iface_data_t *model_data = multinet->create(mn_name, 6000);
    int mu_chunksize = multinet->get_samp_chunksize(model_data);
    esp_mn_commands_update_from_sdkconfig(multinet, model_data); // Add speech commands from sdkconfig
    assert(mu_chunksize == afe_chunksize);
    //print active speech commands
    multinet->print_active_speech_commands(model_data);

    printf("------------detect start------------\n");
    while (task_flag) {
        afe_fetch_result_t* res = afe_handle->fetch(afe_data); 
        if (!res || res->ret_value == ESP_FAIL) {
            printf("fetch error!\n");
            break;
        }

        if (res->wakeup_state == WAKENET_DETECTED) {
            printf("WAKEWORD DETECTED\n");
	    multinet->clean(model_data);
        } else if (res->wakeup_state == WAKENET_CHANNEL_VERIFIED) {
            play_voice = -1;
            detect_flag = 1;
            printf("AFE_FETCH_CHANNEL_VERIFIED, channel index: %d\n", res->trigger_channel_id);
            // afe_handle->disable_wakenet(afe_data);
            // afe_handle->disable_aec(afe_data);
        }

        if (detect_flag == 1) {
            esp_mn_state_t mn_state = multinet->detect(model_data, res->data);

            if (mn_state == ESP_MN_STATE_DETECTING) {
                continue;
            }

            if (mn_state == ESP_MN_STATE_DETECTED) {
                esp_mn_results_t *mn_result = multinet->get_results(model_data);
                for (int i = 0; i < mn_result->num; i++) {
                    printf("TOP %d, command_id: %d, phrase_id: %d, string: %s, prob: %f\n", 
                    i+1, mn_result->command_id[i], mn_result->phrase_id[i], mn_result->string, mn_result->prob[i]);
                }
                printf("-----------listening-----------\n");
            }

            if (mn_state == ESP_MN_STATE_TIMEOUT) {
                esp_mn_results_t *mn_result = multinet->get_results(model_data);
                printf("timeout, string:%s\n", mn_result->string);
                afe_handle->enable_wakenet(afe_data);
                detect_flag = 0;
                printf("\n-----------awaits to be waken up-----------\n");
                continue;
            }
        }
    }
    if (model_data) {
        multinet->destroy(model_data);
        model_data = NULL;
    }
    printf("detect exit\n");
    vTaskDelete(NULL);
}

void app_main(void)
{
    printf("Program start\n--------------------------------------\n");

    init_microphone();
    init_esp_sr();

    task_flag = 1;
    xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void*)afe_data, 5, NULL, 1);
    xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void*)afe_data, 5, NULL, 0);

}
```

feizi, 2023-11-20:
Hi @hornej , Only ESP32-S3 chip support English Multinet.
But your issues is not this reason. I will take a look.

hornej, 2023-11-20:
OK thank you. I am using ESP32-S3

feizi, 2023-11-29:
I do not known how to fix this issue. Your code looks right.
I found `longjmp`  function in flite_g2p module which is a new module in esp-sr. So please update esp-sr(v1.6.0) and try again.

hornej, 2023-11-29:
I just upgraded to esp-sr v1.6 but still the same issue
```
/Users/joshhorne/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/joshhorne/git-dev/esp32-projects/alphabot/managed_components/espressif__esp-sr/lib/esp32s3/libflite_g2p.a(cst_utterance.c.obj):(.literal.utt_relation+0x8): undefined reference to `__wrap_longjmp'
/Users/joshhorne/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/joshhorne/git-dev/esp32-projects/alphabot/managed_components/espressif__esp-sr/lib/esp32s3/libflite_g2p.a(cst_utterance.c.obj): in function `utt_relation':
/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/hrg/cst_utterance.c:108: undefined reference to `__wrap_longjmp'
/Users/joshhorne/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/joshhorne/git-dev/esp32-projects/alphabot/managed_components/espressif__esp-sr/lib/esp32s3/libflite_g2p.a(cst_alloc.c.obj): in function `cst_safe_alloc':
/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_alloc.c:76: undefined reference to `__wrap_longjmp'
/Users/joshhorne/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_alloc.c:115: undefined reference to `__wrap_longjmp'
/Users/joshhorne/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/joshhorne/git-dev/esp32-projects/alphabot/managed_components/espressif__esp-sr/lib/esp32s3/libflite_g2p.a(cst_val.c.obj): in function `val_int':
/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_val.c:146: undefined reference to `__wrap_longjmp'
/Users/joshhorne/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/joshhorne/git-dev/esp32-projects/alphabot/managed_components/espressif__esp-sr/lib/esp32s3/libflite_g2p.a(cst_val.c.obj):/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_val.c:163: more undefined references to `__wrap_longjmp' follow
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

feizi, 2023-11-29:
Could you try to add a line in [CMakeList.txt(line:80)](https://github.com/espressif/esp-sr/blob/master/CMakeLists.txt#L80) of esp-sr
```
add_prebuilt_library(nsnet "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libflite_g2p.a" PRIV_REQUIRES ${COMPONENT_NAME})
```

hornej, 2023-11-29:
OK I added that and moved esp-sr from managed_components to the components folder. I also had to update the partition table from "Single factory app, no OTA" to "Single factory app (large), no OTA" and then I was able to build successfully. 

I am now getting an error where the device is constantly rebooting. If it is unrelated I can figure it out on my own or submit another issue 

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (31) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (31) boot: compile time Nov 28 2023 23:25:16
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00177000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=17010h ( 94224) map
I (121) esp_image: segment 1: paddr=00027038 vaddr=3fc93200 size=04b88h ( 19336) load
I (126) esp_image: segment 2: paddr=0002bbc8 vaddr=40374000 size=04450h ( 17488) load
I (131) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=484c4h (296132) map
I (189) esp_image: segment 4: paddr=000784ec vaddr=40378450 size=0ad5ch ( 44380) load
I (205) boot: Loaded app from partition at offset 0x10000
I (206) boot: Disabling RNG early entropy source...
I (217) cpu_start: Multicore app
I (217) cpu_start: Pro cpu up.
I (217) cpu_start: Starting app cpu, entry point is 0x4037531c
0x4037531c: call_start_cpu1 at /Users/joshhorne/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (236) cpu_start: Pro cpu start user code
I (236) cpu_start: cpu freq: 160000000 Hz
I (236) cpu_start: Application information:
I (239) cpu_start: Project name:     alphabot
I (244) cpu_start: App version:      1
I (248) cpu_start: Compile time:     Nov 28 2023 23:25:07
I (254) cpu_start: ELF file SHA256:  25fa5c2266546979...
I (260) cpu_start: ESP-IDF:          v5.1.1-dirty
I (266) cpu_start: Min chip rev:     v0.0
I (270) cpu_start: Max chip rev:     v0.99 
I (275) cpu_start: Chip rev:         v0.2
I (280) heap_init: Initializing. RAM available for dynamic allocation:
I (287) heap_init: At 3FC98AA0 len 00050C70 (323 KiB): DRAM
I (293) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (300) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (306) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (313) spi_flash: detected chip: gd
I (317) spi_flash: flash io: dio
W (320) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (334) sleep: Configure to isolate all GPIO pins in sleep state
I (341) sleep: Enable automatic switching of GPIO sleep configuration
I (348) app_start: Starting scheduler on CPU0
I (353) app_start: Starting scheduler on CPU1
I (353) main_task: Started on CPU0
I (363) main_task: Calling app_main()
Program start
--------------------------------------
I (373) AFE_SR: afe interface for speech recognition

I (373) AFE_SR: AFE version: SR_V220727

W (383) SR_SYS: CPU freq should be 240MHz
W (383) SR_SYS: PSRAM freq should be not less than 80MHz
W (393) SR_SYS: Data cache recommends 64KB
W (393) SR_SYS: Data cache line recommends 64B
I (403) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (413) AFE_SR: aec_init: 1, se_init: 1, vad_init: 1

I (413) AFE_SR: wakenet_init: 1

Item psram alloc failed. Size: 2068 = 1024 x 2 + 16 + 4
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200bdf9  PS      : 0x00060030  A0      : 0x8200ac84  A1      : 0x3fc9c4e0  
0x4200bdf9: esp_aec3_hps16fft_init at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/acoustic_echo_cancellation/esp_aec3_hps16fft.c:72

A2      : 0x00000000  A3      : 0x00000200  A4      : 0x00000800  A5      : 0x00000003  
A6      : 0x00000000  A7      : 0x00000002  A8      : 0x8200bde4  A9      : 0x00000400  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000404  A13     : 0x00000002  
A14     : 0x00000010  A15     : 0x00000004  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
0x400556d5: strlen in ROM

0x400556e5: strlen in ROM

Backtrace: 0x4200bdf6:0x3fc9c4e0 0x4200ac81:0x3fc9c510 0x42009806:0x3fc9c540 0x4200981c:0x3fc9c5d0 0x42047ceb:0x3fc9c5f0 0x4037d351:0x3fc9c620
0x4200bdf6: esp_aec3_hps16fft_init at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/acoustic_echo_cancellation/esp_aec3_hps16fft.c:75

0x4200ac81: afe_create_from_config at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr.c:268

0x42009806: init_esp_sr at /Users/joshhorne/git-dev/esp32-projects/alphabot/main/main.c:60

0x4200981c: app_main at /Users/joshhorne/git-dev/esp32-projects/alphabot/main/main.c:181

0x42047ceb: main_task at /Users/joshhorne/esp/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x4037d351: vPortTaskWrapper at /Users/joshhorne/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 25fa5c2266546979

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403756e8
0x403756e8: esp_restart_noos_dig at /Users/joshhorne/esp/esp-idf/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)
```

sun-xiangyu, 2023-11-29:
> Item psram alloc failed. Size: 2068 = 1024 x 2 + 16 + 4
> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

From your log, there are not enough PSRAM to alloc. You can refer to [this document](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/benchmark/README.html) to get the memory consumpution of different modules in esp-sr .

closed 2024-04-10 by feizi