espressif/esp-sr #85: AFE_SR: Ns net mode does not supported ESP_AFE_SR_HANDLE!

stintel, 2023-12-13:
We're trying to enable nsnet1 in Willow to see if it can improve the experience in noisy environments. Unfortunately we're hitting the following error:

```
E (5943) AFE_SR: Ns net mode does not supported ESP_AFE_SR_HANDLE!

E (5949) AFE_SR: AFE config error!

E (20:56:22.316) RECORDER_SR: recorder_sr.c:574 (recorder_sr_create): Got NULL Pointer
```

We've disabled AEC as suggested in espressif/esp-skainet@4643ec24076a69c39c9a5bf0d3f2c9eaacbbb4b0, and confirmed that the model is available in the model partition:

```
I (20:56:22.289) WILLOW/AUDIO: Loaded SR models:
* mn6_en
* fst
* wn9_hilexin
* wn9_alexa
* wn9_hiesp
* nsnet1
```

From the error message "Ns net mode does not supported ESP_AFE_SR_HANDLE!" it is not clear why it is failing. Pointers would be appreciated.

shenxiaozheng, 2023-12-14:
Thank you very much for your suggestion. During our experiment, nsnet will cause significant distortion to the speech and reduce the accuracy of speech recognition in noisy scenarios. Therefore, we will only use it for voice call scenarios. We will continuously improve our algorithms, thank you.

stintel, 2023-12-14:
Understood! Might be useful to clarify this in docs and/or release notes.

shenxiaozheng, 2023-12-15:
Okay, thank you very much for your suggestion. I will distinguish between speech recognition and voice calling.

closed 2024-01-09 by feizi