espressif/esp-sr #88: Want to suggest a wake word? Leave your thoughts here.

feizi, 2023-12-14:
Hi all,

We're excited to offer the community more free and high-quality wake word models. Everyone has their own unique wake word preferences. Now, we're ready to regularly release some of the most popular wake words. Please let us know the wake words you want! English and Chinese are both welcome.

In the past, it was an expensive process to collect high-quality human speech data. But now, our team has developed a cost-effective way to train wake word models by using only TTS samples, which reaches 90-95% accuracy compared to models trained by human-recorded samples.

The wake word models and esp-sr have the same license and are free for commercial use. If you want a more accurate and exclusive wake word, please use our [wake word customization service](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/wake_word_engine/ESP_Wake_Words_Customization.html).

kristiankielhofner, 2023-12-14:
The Willow team and community would love "Hey Willow". It's our [domain name](https://heywillow.io/) because we've been waiting for this.

Thank you very much for offering this option, it's very exciting!

feizi, 2023-12-14:
> The Willow team and community would love "Hey Willow". It's our [domain name](https://heywillow.io/) because we've been waiting for this.
> 
> Thank you very much for offering this option, it's very exciting!

I'm glad you like this. Since "hey" and "hi" sound pretty similar, sometimes people might not really notice the difference. So, I was thinking, maybe we could support both "hey willow" and "hi willow" for waking up the device. That way, whether you say "hey willow" or "hi willow", it'll still work. Of course, when we release the wake word model, we'll call it like "wn9_heywillow". What do you think about that?

kristiankielhofner, 2023-12-14:
Good idea!

My only concern would be overall reduced accuracy (wake reliability vs false wake). We've noticed quite a bit of false wake with Alexa. From what I've read the automated TTS approach has 90-95% the accuracy of the models trained on human samples. I like "two word" wake words because they tend to improve accuracy, I suspect a 100% "Hey Willow" wake word could result in equivalent or even improved accuracy with the TTS approach vs even human sample trained Alexa?

Of course we could always test this, even starting with a pure "Hey Willow" model, a pure "Hi Willow" model, and a merged model.

Thanks again for offering this!

feizi, 2023-12-28:
"hey/hi willow" model: **wn9_heywillow_tts** has been added.
FAR(False Alarm Rate): 1 times / 8 hours (test it by 64 hours audio data)

AigizK, 2023-12-29:
Guys, what you are doing is really great. We have created a smart speaker called Homai based on the esp32-s3. We trained the model ourselves, but it is resource-intensive and not so easy to integrate into the pipeline. Could you please add support for our word Homai [ho'mai]? Thank you in advance!

sun-xiangyu, 2024-01-03:
Hi @AigizK ,
The syllable of **Homai** only has two. It is difficult to reduce the probability of false triggering for monosyllabic and disyllabic phrases.  We recommend selecting a 3-5 syllable phrase as the wake word.

AigizK, 2024-01-03:
Hi @sun-xiangyu 
We have already launched a project with this name, so we can't change it significantly. But can we use the variant "homa ai", where the sound 'A' is pronounced long?

sun-xiangyu, 2024-01-04:
> We have already launched a project with this name, so we can't change it significantly. But can we use the variant "homa ai", where the sound 'A' is pronounced long?

I'm sorry that our TTS model cannot specify a syllable to extend its pronunciation at the moment. This means that we cannot generate a large number of accurate “homa ai” phrases.

PrathamG, 2024-01-09:
Hi! Thank you for this awesome solution! We are developing a smart voice assistant called Sophia. Would it be possible to have the wake word "Hi Sophia"? This would help our user experience drastically. Thank you in advance!

sun-xiangyu, 2024-01-09:
Hi @PrathamG , I'm glad you like it. "Sophia" sounds like a wake word that can be used directly. I mean, maybe we don't need an extra prefix "Hi". I suggest we start with just "Sophia". If the performance is not satisfactory, then we can train another one with "hi Sophia". What do you think?

PrathamG, 2024-01-09:
Sure, that sounds like a good plan! We can use only "Sophia" and test the performance first. Thank you

PrathamG, 2024-01-09:
If possible, I also wanted to request the wake word "Little Sophia". We are still unsure about which wake word to use, and having both options will help us determine this via user testing.

sun-xiangyu, 2024-01-10:
> If possible, I also wanted to request the wake word "Little Sophia". We are still unsure about which wake word to use, and having both options will help us determine this via user testing.

Now our computing resources are limited, and this project can generate about two wake word models in a month. So we will choose some popular wake words. Of course, if we have some free time, "Little Sophia" is also fine.

PrathamG, 2024-01-12:
No worries, totally understandable! Looking forward to testing out the "Sophia" wake word

sun-xiangyu, 2024-01-18:
**"Sophia" model: wn9_sophia_tts**

FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 97%

sun-xiangyu, 2024-01-23:
@xygh, “小美同学” sounds good.

PrathamG, 2024-01-23:
> **"Sophia" model: wn9_sophia_tts**
> 
> FAR(False Alarm Rate): 1 times / 8 hours RAR(Right Alarm Rate): 97%

Thank you! We will test it out and report the results by next week

xygh, 2024-01-23:
> @xygh, “小美同学” sounds good.

BTW, “你好小美” is also a perfect choice.

Henry586, 2024-01-25:
"小当家" or "Hi 小星" is preferable wake word in our scenario.  Thanks a lot!

sun-xiangyu, 2024-01-25:
**The second version "Sophia":**
model info: `wakenet9l_tts1h8v2_Sophia_3_0.647_0.649`

**Perfromace:**
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 95%

**Improvement:**
Add "Sophie" and "Sophy" as hard negatives to reduce false triggers.

sun-xiangyu, 2024-01-25:
> "小当家" or "Hi 小星" is preferable wake word in our scenario. Thanks a lot!

Both of these words sound good. If you have no preference, we will choose "hi 小星".

lewardo, 2024-02-11:
Hello! This is a great opportunity I was hoping would come up, I'm so glad this is now possible! I've seen that the wake-words "Mycroft" and "Hey, Mycroft" are very popular in the community, and it is also the name of my product so would very much improve user experience. Would it be possible to have either of these trained and released for the community? Thank you so much in advance for this!

sun-xiangyu, 2024-02-19:
@lewardo,  I'm glad it could help you. Although "Mycroft" is simpler, it seems there are quite a few words that sound similar, so I'll prioritize training with "Hey Mycroft."

sun-xiangyu, 2024-02-19:
@Henry586 , 
Hi,小星
model info: wakenet9l_tts1h8_Hi,小星_3_0.626_0.630

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 93%

jmattsson, 2024-02-27:
I'd love to have "hey printer" available as a wake word/phrase.

Littledragon-wxl, 2024-02-29:
I want to suggest a wake word ,"hey,mybot".
I'm glad to hear that you can create a wake word.

Littledragon-wxl, 2024-02-29:
I want to suggest a wake word ,"小龙小龙".
I'm glad to hear that you can create a wake word.

feizi, 2024-03-04:
@lewardo ,  The performance of "Mycroft" also looks good. Pls try.
Mycroft: wakenet9l_tts1h8_Mycroft_3_0.625_0.629

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 96%

welkinchan, 2024-03-18:
您好！我们正在开发一款名为喵喵同学的智能语音助手。是否帮我们实现一个“喵喵同学”的唤醒词，这将极大地帮助我们提升用户体验。先谢谢您！

sun-xiangyu, 2024-03-18:
Hey,Printer: wakenet9l_tts1h8_Heyprinter_3_0.623_0.629

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 78%

sun-xiangyu, 2024-03-18:
小龙小龙: `wakenet9l_tts1h8_小龙小龙_3_0.624_0.628`

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 95%

jmattsson, 2024-03-19:
@sun-xiangyu Ooh, thank you! I'll try to test Hey Printer on the weekend!

jhbruhn, 2024-03-21:
Is there also a chance to get wakewords trained for the ESP32 variante (WakeNet5 if I understand it correctly)? Or is that obsolete? I would love to use an "Alexa" or "ok nabu" model on my M5Stack Atom Echos, which unfortunately only have an ESP32.

sun-xiangyu, 2024-03-22:
Hi @jhbruhn ,

Yes, WakeNet5 has been deprecated, and we are not training any WakeNet5 models. The ESP32 should be able to run WakeNet9, but we have not yet adapted it. This is because if you want to develop a wake word app with stable performance, it requires running in conjunction with the Audio Front End (AFE). It is diffcult for ESP32 to run in real time. Therefore, we recommend using the ESP32-S3.

Littledragon-wxl, 2024-03-24:
> 小龙小龙: `wakenet9l_tts1h8_小龙小龙_3_0.624_0.628`
> 
> Perfromace: FAR(False Alarm Rate): 1 times / 8 hours RAR(Right Alarm Rate): 95%
thank you ,i will try it.

sun-xiangyu, 2024-03-27:
喵喵同学: wakenet9l_tts1h8_喵喵同学_3_0.644_0.648

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 95%

welkinchan, 2024-03-27:
> 喵喵同学: wakenet9l_tts1h8_喵喵同学_3_0.644_0.648
> 
> Perfromace: FAR(False Alarm Rate): 1 times / 8 hours RAR(Right Alarm Rate): 95%

真是杰出的工作~这对我们来说意义重大！
太感谢您们了！
我们会尽快试用。

welkin988, 2024-03-31:
您好！我们打算做一款玩具，有没有机会帮我们实现一个“hi, Joy”的唤醒词？我们非常喜欢这个唤醒词，非常希望能够在esp上实现。@sun-xiangyu

sun-xiangyu, 2024-04-10:
Hi/Hey, Joy: wakenet9l_tts1h8_Hi,Joy_3_0.631_0.633

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 96%

hundaoluo, 2024-04-14:
> Hi/Hey, Joy: wakenet9l_tts1h8_Hi,Joy_3_0.631_0.633
> 
> Perfromace: FAR(False Alarm Rate): 1 times / 8 hours RAR(Right Alarm Rate): 96%

太感谢您了。
但是能否告知一下我应该怎么进行测试吗？
我使用了最新的的sr-1.7.0组件后，menuconfig里边并没有hijoy的选项。
我不知道应该怎么自定义唤醒词。

sun-xiangyu, 2024-04-15:
因为不是每添加一个唤醒词，都会release 一个新的版本。就是说你下载的版本还没有添加Hi,Joy唤醒词，你可以选择
手动的下载esp-sr master branch, 覆盖之前的esp-sr,就可以找到 Hi,Joy 唤醒词了

welkinchan, 2024-04-18:
> 喵喵同学: wakenet9l_tts1h8_喵喵同学_3_0.644_0.648
> 
> Perfromace: FAR(False Alarm Rate): 1 times / 8 hours RAR(Right Alarm Rate): 95%

@sun-xiangyu 
一：
非常感谢。最近测了下喵喵同学的语音唤醒，发现喵喵跟同学两个词要在说的时候明显分开，唤醒成功率才会比较高。如果连在一起效果会差不少。
请问是否有机再帮忙实现下“Hi，喵喵"这个唤醒词？我觉得这个唤醒词的效果应该会好很多。

二：
另外，提供一个建议哈，就是使用TTS语音训练的唤醒词模型，我发现这类模型对发音的准确度要求很高，
大概率是因为TTS的语音的发音都很标准，所以导致训练出来的模型也需要很高的发音准确度才能唤醒。
但是一般人的发音的准确度没有那么高，所以导致训练出来的模型语音识别率并不高。

我在测试sophia以及hi, joy的时候，这个感觉很明显，只有发音非常准的时候才能成功识别，发音差一点就识别失败。
所以我提供的建议是，对于有一些发音比较难的词，如果用TTS训练的话，
可以考虑在训练素材里边增加一些的近音词来一同训练，这样应该可以大大提高识别率。
不知道我对TTS训练的思路的理解对不对哈，仅供参考。

sun-xiangyu, 2024-04-18:
@welkinchan 
非常感谢你的反馈。

1. 可以再训练 “Hi, 喵喵” 这个唤醒词
2. 在唤醒的准确度和区分度上，我们需要去做一个权衡。 比如对于Sophia， 我们特意添加了近义词作为他的负样本来防止他被近义词唤醒。如果不是特别的需求，比如口音和方言等问题，我们一般不会添加近义词，因为这会在某种程度提高模型的误唤醒率。

brainstorm, 2024-04-23:
We have a couple of 3D printers @CCHS-Melbourne called like this:

Wanda
Cosmo
Hey Wanda
Hey Cosmo

Would it be possible to train it ourselves? Is there documentation on how to train with i.e a H100 or Nvidia 4090?

/cc @adricl

feizi, 2024-04-24:
> Would it be possible to train it ourselves? Is there documentation on how to train with i.e a H100 or Nvidia 4090?

We can help you train some  wake words, but our training process isn't open-sourced yet.

dnambinh, 2024-04-26:
@feizi hello, can you train with the model "hey, Li Li" or "Hi, Li Li" or simply "Li Li", I'm a bit hesitant thinking about LiLi vs Li Li, whether whether it makes a big impact on accuracy. I tested this phrase with multinet7 and the results were also very positive, but I lacked a tool to evaluate accuracy.

feizi, 2024-04-28:
Hi @dnambinh 

 For humans, "Li Li" and "LiLi" should be the same, but for TTS (Text-to-Speech), "Li Li" might insert a brief pause.
 I'm not sure how you pronounce "LiLi", whether it's like the English word "Lily" or the Chinese name "莉莉".

dnambinh, 2024-04-28:
> Hi @dnambinh
> 
> For humans, "Li Li" and "LiLi" should be the same, but for TTS (Text-to-Speech), "Li Li" might insert a brief pause. I'm not sure how you pronounce "LiLi", whether it's like the English word "Lily" or the Chinese name "莉莉".

Yes, I have tried common tts tools and there is not much difference between "lily" and "lili". Ya, if it were a word that made sense in English it would be "lily". The initial idea was a certain wake-word that most people (speaking English - Chinese - Vietnamese - Arabic - ....) could easily say due to similarities in pronunciation. I'd be happy if you could train a similar model

sun-xiangyu, 2024-04-28:
> Yes, I have tried common tts tools and there is not much difference between "lily" and "lili". Ya, if it were a word that made sense in English it would be "lily". The initial idea was a certain [...]

OK，I will use both "Hi,Lily" and "Hi, 丽丽" to train a wake word model.

sun-xiangyu, 2024-05-08:
Hi,Lily/Hi,莉莉: wakenet9l_tts1h8_Hi,Lily or Hi,莉莉_3_0.633_0.639

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 93%

@dnambinh , Looking forward to your feedback

dnambinh, 2024-05-08:
> Hi,Lily/Hi,莉莉: wakenet9l_tts1h8_Hi,Lily or Hi,莉莉_3_0.633_0.639
> 
> Perfromace: FAR(False Alarm Rate): 1 times / 8 hours RAR(Right Alarm Rate): 93%
> 
> @dnambinh , looking forward to your feedback

Thanks for your support, i will feedback asap

usama1123456789, 2024-05-12:
An arabic wake word is missing . It could be any arabic name , or the greetings "Assalam-o-alaikum"

sun-xiangyu, 2024-05-14:
Hi @usama1123456789 ,

Now we only can generate English or Chinese TTS samples. we can try to train some wake words like "Hi,Sarah" or "Yasmeen"  by English TTS samples.  But I'm not sure if those model can work well for Arabic.

feizi, 2024-05-14:
Hey,Wanda: wakenet9l_tts1h8_Hey,Wanda_3_0.641_0.644

Perfromace:
FAR(False Alarm Rate): 1 times / 8 hours
RAR(Right Alarm Rate): 95%

@brainstorm , please try.

kristiankielhofner, 2024-05-14:
> An arabic wake word is missing . It could be any arabic name like "Sarah", "Rahma" or "Yasmeen" , or the greetings "Assalam-o-alaikum"

From what I've seen use of this greeting is extremely common and it's likely you will get a lot of false-wake incidents when it is so frequently used in everyday conversation within audio range of your devices.

I'd suggest you recommend an acceptable alternative.

usama1123456789, 2024-05-14:
> From what I've seen use of this greeting is extremely common and it's likely you will get a lot of false-wake incidents when it is so frequently used in everyday conversation within audio range of [...]

You are correct! that greeting would invoke many unwanted wakeups.

By the way can we get one called Hi, Astrolabe.?  in english ofcourse

kaylyun, 2024-05-15:
We are making cultural and creative products with a Harbin theme, and we hope to have some 'awakening words' like 'Hey, Xiaobin' or 'Xiaobin Xiaobin'. Could you please help us train that? Thank you.
@sun-xiangyu

sun-xiangyu, 2024-05-16:
> We have a device called Astrolabe and would like to get one model trained for that word!

@usama1123456789 , Astrolabe sounds complex enough to be a stand-alone wake word, and I recommend leaving it unprefixed with "Hi"

dnambinh, 2024-05-16:
> Hi,Lily/Hi,莉莉: wakenet9l_tts1h8_Hi,Lily or Hi,莉莉_3_0.633_0.639
> 
> Perfromace: FAR(False Alarm Rate): 1 times / 8 hours RAR(Right Alarm Rate): 93%
> 
> @dnambinh , looking forward to your feedback

Hi, @sun-xiangyu. I tried model 'Hi,Lily'. The results received were extremely positive.
Perform:
+) FAR(False Alarm Rate): Compared to the 'Alexa' model, Alexa's false activation rate is clearly higher, simply by putting the device into a random conversation.
+) RAR: When I knew that the model using TTS was only about 80-90% effective compared to the model using real human voice, and RAR was about 93%, I didn't expect much. However, the result was surprising, it was surprisingly accurate, I tried with the accent of almost everyone in the company, with other nationalities, even with the local accent, which is extremely difficult to hear.

From here it can be concluded that choosing the correct wakeword is extremely important and greatly affects the results.
Test device configuration:
+) ESP32S3
+) 1MIC vs 2MIC mems

However, I'm having a new problem. The idea is to wake up the device with a wakeword, then record until the VAD checks that there is no more voice (I checked for 1 second) and then send it to the server for further processing. I noticed that the VAD cannot check the sound of people far away (2m), while the wakeword still works well. If I reduce the VAD level to 2 or 1 it gets better however it is easily triggered by other noise.
Is there any feasible solution?

usama1123456789, 2024-05-16:
> > We have a device called Astrolabe and would like to get one model trained for that word!
> 
> @usama1123456789 , Astrolabe sounds complex enough to be a stand-alone wake word, and I recommend leaving it unprefixed with "Hi"

ASTROLABE as a standalone would be OK  and acceptable. Can we get that ?

[341 more replies]