espressif/esp-sr #89: feature request: ability to check SR model version

stintel, 2024-01-09:
First of all I'd like to thank you for the recently added new wake words. I've been testing Hey Willow on Willow for almost two weeks now, and so far the experience has been very positive.

We plan to make the new wake words available in Willow in our next release. As we don't want to ask our users to do a full flash, which would require them to collect all their devices from wherever they've deployed them in their houses, we're working on a feature to facilitate OTA updates of the srmodel partition. We actually already have a working PoC to do this. The PoC currently unconditionally flashes the model partition. However, it would be nice if we can query the version of the wake model(s) currently flashed on the device. This way, we can avoid flashing the model partition when it is not needed.

Would you consider adding a field to the srmodel_data_t struct to make this possible?

feizi, 2024-01-10:
OK, we will add one.

sun-xiangyu, 2024-01-16:
Hi @stintel,
Now we can use `model_info` string in `srmodel_data_t` to represent the wakenet version. Please refer to[ this line](https://github.com/espressif/esp-sr/blob/master/src/include/model_path.h#L21).

closed 2024-01-30 by feizi