espressif/esp-sr #90: feature request: add friendly name to WakeNet models

stintel, 2024-01-09:
In Willow, we set a wake help text on the display to inform the user which wake word to use, e.g. `Say 'Alexa' to start`. We currently hardcode the friendly name [in the C code](https://github.com/toverainc/willow/blob/74dfff24a1173c5c393400fd93c63e0edc7593bb/main/audio.c#L972-L984). This works fine for the 3 wake words we currently support. In the future, we would like to offer our users a way to flash new models as they become available, without having to update the Willow C code, so this would no longer work.

Would you consider adding a field to the WakeNet models for such a friendly name, so we can remove the hardcoded friendly names from our C code? Thanks!

feizi, 2024-01-10:
This idea seem quite reasonable, we will add one.

sun-xiangyu, 2024-01-16:
The friendly name was defined in [_MODEL_INFO_ ](https://github.com/espressif/esp-sr/blob/master/model/wakenet_model/wn9_heywillow_tts/_MODEL_INFO_) file.
 We add [esp_srmodel_get_wake_words](https://github.com/espressif/esp-sr/blob/master/src/include/model_path.h#L77C1-L85C76) function to parse friendly name from `model_info`.  If there are multiple wake words in one model, the wake words will be joined by ";". 

You can use it like this:
```
srmodel_list_t *models = esp_srmodel_init("model");
char *model_name = esp_srmodel_filter(models, ESP_WN_PREFIX, NULL);
char *wake_words = wake_words = esp_srmodel_get_wake_words(models, model_name );
```

closed 2024-01-30 by feizi