espressif/esp-sr #91: Suitable for non-human sound recognition?

gamename, 2024-02-22:
Hi,

Is it possible to use this package for identifying non-human sounds? I'm looking for a way to recognize cat sounds.  I have the model, but I don't know what ESP library to use.

Any suggestions would be appreciated.

shenxiaozheng, 2024-02-23:
I understand that you need to trigger a device when a cat makes a sound, similar to Amazon's smart speakers with the wake-up word "Alexa." However, AFE does not have the wake-up word "meow," so you hope to provide an audio sample of a cat meowing for wake word model training.

gamename, 2024-02-23:
> > I have the model, but I don't know what ESP library to use.
> 
> Hi @gamename , If you want to deploy your own model on ESP chip, please refer to [esp-dl](https://github.com/espressif/esp-dl). This project may help you.

Thank you!

closed 2024-03-12 by gamename