espressif/esp-sr #92: Building error when used in platformIO/ESPHome context

X-Ryl669, 2024-03-11:
When used in ESPHome context (for Home Assistant mainly), any version later than [this commit](https://github.com/espressif/esp-sr/commit/e1c73a59d94fe5d656c83788c9cec619e8f5070a) fails CMake to build on this line
`idf_component_get_property(esp_dsp_lib espressif__esp-dsp COMPONENT_LIB)`

This is because `espressif__esp-dsp` isn't defined so CMake choke on bad number of argument to the `idf_component_get_property`, it see 2 arguments instead of 3.

I do have a `__idf_esp-dsp` variable, but not a `espressif__esp-dsp` variable. I'm able to successfully pass this step if I just replace that line with `set(esp_dsp_lib "esp-dsp")`

X-Ryl669, 2024-03-11:
@igrr I don't know how the `espressif__esp-dsp` is created (maybe there's an additional step to perform here that I didn't). Do you have an idea?

igrr, 2024-03-11:
Could you please describe how you add esp-sr to your project?

Perhaps you could also attach the build log of a clean (failing) build, as well as the (successful) build log after the modification you made?

closed 2024-03-11 by X-Ryl669