espressif/esp-sr #93: Voice Activity Detection (VAD) and Automatic Gain Control (AGC) functionality simultaneously

PrathamG, 2024-03-13:
Hi! If I understand correctly, we can currently only use VAD with ESP_AFE_SR_HANDLE and AGC with ESP_AFE_VC_HANDLE. 
I have a use case where I want to first carry out VAD on the input stream, and if VAD is true, carry out AGC for some data processing. 
Is it possible to implement AGC and VAD at the same time? Thanks!

sun-xiangyu, 2024-03-13:
Hi @PrathamG ,
The AGC of ESP_AFE_SR_HANDLE is enabled by default. You can modify agc_mode to control the output gain. 
Please refer to [afe_mn_peak_agc_mode_t agc_mode](https://github.com/espressif/esp-sr/blob/50893a21af2fa18e63f30ec297f8788c71bcef1d/include/esp32s3/esp_afe_config.h#L87) for more details.

PrathamG, 2024-03-13:
Thanks @sun-xiangyu ! 
So currently it's not possible to use the AGC of ESP_AFE_SR_HANDLE without enabling the wakenet, is that correct?

sun-xiangyu, 2024-03-13:
Yes, you only can set a fix gain by[ float afe_linear_gain](https://github.com/espressif/esp-sr/blob/50893a21af2fa18e63f30ec297f8788c71bcef1d/include/esp32s3/esp_afe_config.h#L85).