espressif/esp-sr #95: ESP-SR failed during linking.  undefined references to `__wrap_longjmp' follow

nikhil-robinson, 2024-04-10:
Trying out the english speech detetction.

IDF version 5.2.1

Target esp32s3 box 3

```
FAILED: kunadava-ai.elf
cmd.exe /C "cd . && C:\Users\nikhi\esp\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/nikhi/Desktop/Nikhil/kundappa-ai/build/kunadava-ai.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\kunadava-ai.elf.rsp -o kunadava-ai.elf  && cd ."        
C:/Users/nikhi/esp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/nikhi/Desktop/Nikhil/kundappa-ai/components/esp-sr/lib/esp32s3\libflite_g2p.a(cst_utterance.c.obj):(.literal.utt_relation+0x8): undefined reference to `__wrap_longjmp'
C:/Users/nikhi/esp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/nikhi/Desktop/Nikhil/kundappa-ai/components/esp-sr/lib/esp32s3\libflite_g2p.a(cst_utterance.c.obj): in function `utt_relation':
/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/hrg/cst_utterance.c:108:(.text.utt_relation+0x1e): undefined reference to `__wrap_longjmp'
C:/Users/nikhi/esp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/nikhi/Desktop/Nikhil/kundappa-ai/components/esp-sr/lib/esp32s3\libflite_g2p.a(cst_alloc.c.obj): in function `cst_safe_alloc':
/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_alloc.c:76:(.text.cst_safe_alloc+0x18): undefined reference to `__wrap_longjmp'
C:/Users/nikhi/esp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: /home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_alloc.c:115:(.text.cst_safe_alloc+0x4a): undefined reference to `__wrap_longjmp'
C:/Users/nikhi/esp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/nikhi/Desktop/Nikhil/kundappa-ai/components/esp-sr/lib/esp32s3\libflite_g2p.a(cst_val.c.obj): in function `val_int':       
/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_val.c:146:(.text.val_int+0x52): undefined reference to `__wrap_longjmp'   
C:/Users/nikhi/esp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/nikhi/Desktop/Nikhil/kundappa-ai/components/esp-sr/lib/esp32s3\libflite_g2p.a(cst_val.c.obj):/home/sunxiangyu/workspace/esp_sr_lib/components/flite_g2p/src/utils/cst_val.c:163: more undefined references to `__wrap_longjmp' follow
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1. 
```

code :
```

#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_wn_iface.h"
#include "esp_wn_models.h"
#include "esp_afe_sr_iface.h"
#include "esp_afe_sr_models.h"
#include "esp_mn_iface.h"
#include "esp_mn_models.h"
#include "model_path.h"
#include "esp_process_sdkconfig.h"
#include "esp_mn_speech_commands.h"

#include "esp_log.h"
#include "bsp/esp-bsp.h"

int detect_flag = 0;
static esp_afe_sr_iface_t *afe_handle = NULL;
static volatile int task_flag = 0;
srmodel_list_t *models = NULL;
static int play_voice = -2;

void play_music(void *arg)
{
    while (task_flag) {
        switch (play_voice) {
        case -2:
            vTaskDelay(10);
            break;
        case -1:
            wake_up_action();
            play_voice = -2;
            break;
        default:
            speech_commands_action(play_voice);
            play_voice = -2;
            break;
        }
    }
    vTaskDelete(NULL);
}

void feed_Task(void *arg)
{
   ESP_LOGI(TAG, "Feed Task");
    size_t bytes_read = 0;
    esp_afe_sr_data_t *afe_data = (esp_afe_sr_data_t *) arg;
    int audio_chunksize = afe_handle->get_feed_chunksize(afe_data);
    int feed_channel = 3;
    ESP_LOGI(TAG, "audio_chunksize=%d, feed_channel=%d", audio_chunksize, feed_channel);

    int16_t *audio_buffer = heap_caps_malloc(audio_chunksize * sizeof(int16_t) * feed_channel, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
    assert(audio_buffer);

    while (true) {

        ESP_ERROR_CHECK(esp_codec_dev_read(mic_handel, audio_buffer, audio_chunksize * I2S_CHANNEL_NUM * sizeof(int16_t)));

        for (int  i = audio_chunksize - 1; i >= 0; i--) {
            audio_buffer[i * 3 + 2] = 0;
            audio_buffer[i * 3 + 1] = audio_buffer[i * 2 + 1];
            audio_buffer[i * 3 + 0] = audio_buffer[i * 2 + 0];
        } 

        afe_handle->feed(afe_data, audio_buffer);
    }
    if (audio_buffer) {
        free(audio_buffer);
        audio_buffer = NULL;
    }
    vTaskDelete(NULL);
}

void detect_Task(void *arg)
{
    esp_afe_sr_data_t *afe_data = arg;
    int afe_chunksize = afe_handle->get_fetch_chunksize(afe_data);
    char *mn_name = esp_srmodel_filter(models, ESP_MN_PREFIX, ESP_MN_ENGLISH);
    printf("multinet:%s\n", mn_name);
    esp_mn_iface_t *multinet = esp_mn_handle_from_name(mn_name);
    model_iface_data_t *model_data = multinet->create(mn_name, 6000);
    int mu_chunksize = multinet->get_samp_chunksize(model_data);
    esp_mn_commands_update_from_sdkconfig(multinet, model_data); // Add speech commands from sdkconfig
    assert(mu_chunksize == afe_chunksize);
    //print active speech commands
    multinet->print_active_speech_commands(model_data);

    printf("------------detect start------------\n");
    while (task_flag) {
        afe_fetch_result_t* res = afe_handle->fetch(afe_data); 
        if (!res || res->ret_value == ESP_FAIL) {
            printf("fetch error!\n");
            break;
        }

        if (res->wakeup_state == WAKENET_DETECTED) {
            printf("WAKEWORD DETECTED\n");
	    multinet->clean(model_data);
        } else if (res->wakeup_state == WAKENET_CHANNEL_VERIFIED) {
            play_voice = -1;
            detect_flag = 1;
            printf("AFE_FETCH_CHANNEL_VERIFIED, channel index: %d\n", res->trigger_channel_id);
            // afe_handle->disable_wakenet(afe_data);
            // afe_handle->disable_aec(afe_data);
        }

        if (detect_flag == 1) {
            esp_mn_state_t mn_state = multinet->detect(model_data, res->data);

            if (mn_state == ESP_MN_STATE_DETECTING) {
                continue;
            }

            if (mn_state == ESP_MN_STATE_DETECTED) {
                esp_mn_results_t *mn_result = multinet->get_results(model_data);
                for (int i = 0; i < mn_result->num; i++) {
                    printf("TOP %d, command_id: %d, phrase_id: %d, string: %s, prob: %f\n", 
                    i+1, mn_result->command_id[i], mn_result->phrase_id[i], mn_result->string, mn_result->prob[i]);
                }
                printf("-----------listening-----------\n");
            }

            if (mn_state == ESP_MN_STATE_TIMEOUT) {
                esp_mn_results_t *mn_result = multinet->get_results(model_data);
                printf("timeout, string:%s\n", mn_result->string);
                afe_handle->enable_wakenet(afe_data);
                detect_flag = 0;
                printf("\n-----------awaits to be waken up-----------\n");
                continue;
            }
        }
    }
    if (model_data) {
        multinet->destroy(model_data);
        model_data = NULL;
    }
    printf("detect exit\n");
    vTaskDelete(NULL);
}

void app_main()
{
    mic_handel = bsp_audio_codec_microphone_init();

    models = esp_srmodel_init("model"); // partition label defined in partitions.csv
    afe_handle = (esp_afe_sr_iface_t *)&ESP_AFE_SR_HANDLE;

    afe_config_t afe_config = AFE_CONFIG_DEFAULT();
    afe_config.wakenet_model_name = esp_srmodel_filter(models, ESP_WN_PREFIX, NULL);;
    afe_config.aec_init = false;
    esp_afe_sr_data_t *afe_data = afe_handle->create_from_config(&afe_config);
    xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void*)afe_data, 5, NULL, 1);
    xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void*)afe_data, 5, NULL, 0);
}

SDKCONFIG
[sdkconfig.txt](https://github.com/espressif/esp-sr/files/14935313/sdkconfig.txt)

```

feizi, 2024-04-12:
Please try to add the following line in [CMakeLists.txt](https://github.com/espressif/esp-sr/blob/master/CMakeLists.txt#L74)
```
add_prebuilt_library( "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/flite_g2p.a" PRIV_REQUIRES ${COMPONENT_NAME})
```

nikhil-robinson, 2024-04-12:
> Please try to add the following line in [CMakeLists.txt](https://github.com/espressif/esp-sr/blob/master/CMakeLists.txt#L74)
> 
> ```
> add_prebuilt_library( "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/flite_g2p.a" PRIV_REQUIRES ${COMPONENT_NAME})
> ```

worked.

closed 2024-04-12 by nikhil-robinson