espressif/esp-sr #97: More commands (or number recognition)

ftab, 2024-05-15:
I'd like to make a speech recognition interface for a car radio receiver. I think tuning the radio by voice would be cool.

However, according to the readme, multinet only supports up to 300 commands. Depending on the region, there might be 100-200 possible stations for each radio band. This could easily exceed the command limit after all other commands are taken into account.

Is there a way to make generic number recognition so that the user could say "1610" ("sixteen ten"), "98.9" ("ninety eight point nine"), or "3" without me explicitly defining them as commands?

If not, is there a way to increase the number of commands supported in multinet so I could have each station set up as a voice command? What would be the drawbacks of having more than 300 commands?

sun-xiangyu, 2024-05-17:
There is no limit to the number of command words when using the latest multinet programs, we recommend less than 300 because when the number of command words is too large, it is easy to accidentally wake up to other commands

ftab, 2024-05-23:
Thanks for the info! That answers my question.

closed 2024-05-23 by ftab