espressif/esp-sr #98: NSNET ESP-SR

dnambinh, 2024-05-17:
what is different between NSNET1 vs NSNET2?(noise filtering performance and additional features (output amplification?)?

shenxiaozheng, 2024-05-17:
> what is different between NSNET1 vs NSNET2?(noise filtering performance and additional features (output amplification?)?

Compared to nsnet1, nsnet2 has a moderate reduction in CPU usage, but an increase in MOS value of 16%. It achieves better noise reduction ability at lower CPU usage.

shenxiaozheng, 2024-05-17:
Nsnet2 has much lower CPU usage than nsnet1, but its MOS value has increased by 16%. It achieves better noise reduction ability at lower CPU usage.

dnambinh, 2024-05-23:
> Nsnet2 has much lower CPU usage than nsnet1, but its MOS value has increased by 16%. It achieves better noise reduction ability at lower CPU usage.

I have not tested the performance of CPU or RAM usage, but it is clear that when using 2 mic mems, nsnet1 is more effective, nsnet2 causes sound distortion. I wonder if there is any difference in setting up afe or i2s between these two networks?

shenxiaozheng, 2024-05-23:
My testing standard is the MOS value, which is an internationally recognized standard, Nsnet2 has better performance. How did you test for distortion? There is no difference in i2s.

dnambinh, 2024-05-23:
> My testing standard is the MOS value, which is an internationally recognized standard, Nsnet2 has better performance. How did you test for distortion? There is no difference in i2s.

Simply listen and compare with input. Maybe I'll test more with our speech to text model, if it translates then there's no problem. As an additional note, using NS_MODE_SSP mode is also quite good, but I still want to improve the input quality.
I will come back after completing the test. Thank for your infor

shenxiaozheng, 2024-05-23:
If you are using it for a very simple scenario, I highly recommend using SSP mode.

dnambinh, 2024-05-30:
Currently I use MODE SSP and everything is relatively fine so I will close this issue, thank you very much

closed 2024-05-30 by dnambinh

shenxiaozheng, 2024-05-31:
Thank you very much for using our library and for your valuable suggestions. We will continue to optimize and hope to bring you a better experience.