espressif/esp-sr #101: Wake up using API

mike-2020, 2024-07-13:
Instead of letting user speak the wake word, I need to wake up the wakenet in code, so that user can directly speak command word without speaking wake word temporarily. Is there such kind of API?

feizi, 2024-07-15:
Speech command recognition and wake word recognition are independent of each other. I think you can trigger command recognition at any time, regardless of whether it has been woken up.

mike-2020, 2024-07-18:
yes, it works perfect.

Is there a way to control timeout at runtime? I mean, after multinet is created, modify the timeout value ...

feizi, 2024-07-18:
> Is there a way to control timeout at runtime? I mean, after multinet is created, modify the timeout value ...

There is no such API yet

closed 2024-08-31 by mike-2020