espressif/esp-sr #102: Wake-up problem

Sail-211010, 2024-07-16:
[image]
Every time I wake up，If it is the first time to awaken, he must be in channel 2, and subsequent awakenings will be in channel 1，When in channel 2, the command word cannot be recognized, I want to know what to do, either do not enter channel 2 or make it recognize the command word in channel 2

sun-xiangyu, 2024-07-18:
@Sail-211010 ， Thank you for your feedback， I will test it.

Sail-211010, 2024-07-19:
Hi. I wonder if you've already tested it. I'm sorry, every time I try to reply to you, I get interrupted by other things, and then I forget to reply

Sail-211010, 2024-07-23:
@sun-xiangyu Hello, Did you get the results?

feizi, 2024-07-24:
Hi @Sail-211010 , I wander if your wakenet_mode is DET_MODE_3CH_90 or DET_MODE_3CH_95?

Sail-211010, 2024-07-24:
hello @feizi  What do I need to do in ArduinoIDE environment? I can't find the definition in the library file

feizi, 2024-07-24:
It is a config item in afe_config. You can find afe_config in your code and change it.
```
afe_config_t afe_config = AFE_CONFIG_DEFAULT();
afe_config.wakenet_mode = DET_MODE_3CH_90;
```

Sail-211010, 2024-07-24:
@feizi  
[image]
I can't find this parameter,

Sail-211010, 2024-07-31:
[image]
I have modified it, but it doesn't solve my problem
I have tried DET_MODE_3CH_90 and DET_MODE_3CH_95 and they are also consistent with this parameter not configured

Sail-211010, 2024-07-31:
@feizi 
And it does not appear randomly, the first wake up after each power-on is channel 2, and the subsequent channel 2 will not appear

feizi, 2024-08-02:
@Sail-211010 , Yes, by default, the first wake-up uses channel 2, which is the raw data from the microphone. Channels 0 and 1 are the data enhanced by BSS (Blind Source Separation). The reason for this choice is to consider that BSS may not have converged during the first wake-up, which could lead to the selection of the wrong channel, so the default is to choose the original data channel.

Sail-211010, 2024-08-02:
Hi @feizi 
How do I turn this off? I don't want channel 2 to happen, at least I need to turn it off first and test the actual effect

feizi, 2024-08-02:
There is currently no option to turn off this setting; I need to regenerate a new version.

Sail-211010, 2024-08-02:
If possible, modify whether the function model and wakeword model of the English and Chinese models can be placed in a location that can be called
[image]

Sail-211010, 2024-08-02:
Of course this is a different operation,This is a recommendation if you have a version update in this area in the future
My English is not very good. Please don't mind if my presentation is not very good

In the current version, I need to enter the library file to modify these contents, whether there is a chance to solve this operation by calling some functions in the main program in the subsequent version

> There is currently no option to turn off this setting; I need to regenerate a new version.

Sail-211010, 2024-08-02:
Above I am referring to the recommended update for the new firmware version (if you will be making this change).
Thank you very much for your reply
@feizi

feizi, 2024-08-06:
Please try new master branch.
You can set [`afe_config.fixed_first_channel=false`](https://github.com/espressif/esp-sr/blob/02df5f038404417c50e964b22fbc32a7c5292496/include/esp32s3/esp_afe_config.h#L93C10-L93C29) to disable fixed channel number for first wake-up.

Sail-211010, 2024-08-06:
Okay, I'll try. Thanks

Sail-211010, 2024-08-07:
> [`afe_config.fixed_first_channel=false`](https://github.com/espressif/esp-sr/blob/02df5f038404417c50e964b22fbc32a7c5292496/include/esp32s3/esp_afe_config.h#L93C10-L93C29) t
@feizi 
hi In which version is fixed_first_channel started? I can't configure this parameter in the current file, it shows no member

sun-xiangyu, 2024-08-08:
The latest version is v1.9.0

sun-xiangyu, 2024-08-08:
I think you need to update esp-sr manually by github or [Esp Components Registry](https://components.espressif.com/components/espressif/esp-sr).

Sail-211010, 2024-08-08:
> I think you need to update esp-sr manually by github or [Esp Components Registry](https://components.espressif.com/components/espressif/esp-sr).
I understand. Thank you very much for your help. I'll test the feasibility later

Sail-211010, 2024-08-10:
@feizi 

Hello, I looked at this library I want to use on Arduino. How do I operate it? Sensory manipulation can be complicated, okay?

feizi, 2024-08-12:
Hi @Sail-211010 ,
This is [an example](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP_SR) in arduino-esp32 about how to use esp-sr on Arduino.

Sail-211010, 2024-08-12:
@feizi 
[image]
Yes, I know this example, but I need to use the version of v1.9.0. Do you mean that this version is already used on 3.0.4

feizi, 2024-08-12:
[Here ]( https://github.com/espressif/arduino-esp32/blob/master/idf_component.yml#L80) is the version of esp-sr in arduino-esp32.
Arduino-esp32 has their own release plan and priority. If you want to use the latest esp-sr, I suggest you modify it manually.

Sail-211010, 2024-08-12:
@feizi 
[image]

Is there a tutorial for this modification,I'm not sure what needs to be changed or do I just need to change this version?

feizi, 2024-08-12:
I think you can refer to [idf-component-manager ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-component-manager.html).
In my experience, you just need to change the name of the version number and recompile. It will usually be downloaded to **managed_components/espressif_esp-sr**. If you find this folder, open the idf_component.yml file in this folder and make sure the version is v1.9.0. If not, you can clone esp-sr and copy it into this folder by manually.

closed 2024-10-08 by feizi