espressif/esp-sr #105: Ringbuffer Creation issue

ZawarKhan97, 2024-07-20:
Hi, I testing Voice Communication example on esp32s3devkitc1. When during configuration load it gives me error ring buffer not created memory extenuated. Even i increase or reduce ring buffer size afe_config.afe_ringbuf_size = 100.. Help would be appreciated

ZawarKhan97, 2024-07-20:
when i use this afe_handle->reset_buffer(&afe_config);  before afe_data = afe_handle->create_from_config(&afe_config);..It does not give me memory exhaustion error..But it still goes to error of guru mediation error core 0 Paniced

ZawarKhan97, 2024-07-20:
@sun-xiangyu  Kindly help..sorry for my manners

feizi, 2024-07-22:
Please print your memory, including Internal ram and psram. I suspect that the remaining free memory is not enough.

ZawarKhan97, 2024-07-22:
@feizi  thanks. Memory was the issue. I found out the spiram config wasnt done. I fixed it by enabling it in menuconfig. However i still get rb_out slow error. I will try to resolve it  I am not good that much ..

feizi, 2024-07-24:
Hi @ZawarKhan97 , I suggest you use the default kconfig in [examples ](https://github.com/espressif/esp-skainet/blob/master/examples/en_speech_commands_recognition/sdkconfig.defaults.esp32s3) of esp-skainet.

ZawarKhan97, 2024-07-24:
Thanks I have done atlast completed VC example with i2s read mic and i2s speaker. The issues resolved when i configured PSRAM and partition size to custom in menuconfig..

feizi, 2024-07-24:
The configuration of PSRAM affects the read and write speed of PSRAM, which will significantly affect the speed of the esp-sr.
If the memory is slow or the CPU load is heavy, some audio frames may be dropped. Sometimes the audio quality will sound worse.

ZawarKhan97, 2024-07-26:
@feizi  Thanksalot.. I understood the concept. And have successfully implemented Voice communication audio loop back from I2s mic to I2s speaker with AEC,VAD, NS, and AGC enabled. It works pretty good on EP32S3DevkitC1 . Thanks for your contribution.

closed 2024-07-26 by ZawarKhan97