espressif/esp-sr #110: 请问i2s_stream播放http连接是否会导致sr语音唤醒失效？

welkinchan, 2024-08-06:
我最近在尝试开发语音对话。
当我使用audio_player_play从http连接缓存下来的语音文件时，语音唤醒可以正常工作。
但是，其他代码都不改动的情况下，使用i2s_stream的pipeline播放http语音文件链接后，就无法再唤醒设备了。
所以我想确认一下，是不是i2s_stream的pipeline播放http语音链接后，会导致sr语音唤醒失效？

附件是我的audio_player_play的log，audio_player_play的播放代码
i2s_stream的pipeline播放log，以及i2s_stream的pipeline播放代码，

从我个人分析来看，当使用i2s的pipeline进行tts语音的播放后，audio_feed_task 以及audio_detect_task 都是正常在执行的。
但是i2s_stream的pipeline只要执行一遍，就无法唤醒。

audio_player_play的代码以及log
[tts_api.txt](https://github.com/user-attachments/files/16504419/tts_api.txt)
[http tts可以多次唤醒.txt](https://github.com/user-attachments/files/16487718/http.tts.txt)

i2s_stream的pipeline播放代码以及log
[i2s tts第二次无法唤醒.txt](https://github.com/user-attachments/files/16487719/i2s.tts.txt)
[i2s_stream tts代码.txt](https://github.com/user-attachments/files/16487720/i2s_stream.tts.txt)

shootao, 2024-08-09:
@welkinchan 根据你的现象， 像是使用 i2s 出现了问题， 如果使用了 i2s_stream 的话， 就建议使用 i2s stream 作为数据来源了，可以参考如下
```
// First need initializsed i2s audio element as input
static audio_element_handle_t i2s_reader;
void feed_Task(void *arg)
{
    esp_afe_sr_data_t *afe_data = arg;
    int audio_chunksize = afe_handle->get_feed_chunksize(afe_data);
    int nch = afe_handle->get_channel_num(afe_data);
    int feed_channel = esp_get_feed_channel();
    assert(nch <= feed_channel);
    int16_t *i2s_buff = malloc(audio_chunksize * sizeof(int16_t) * feed_channel);
    assert(i2s_buff);

    while (task_flag) {
        audio_element_input(i2s_reader, i2s_buff, audio_chunksize * sizeof(int16_t) * feed_channel)
        afe_handle->feed(afe_data, i2s_buff);
    }
    if (i2s_buff) {
        free(i2s_buff);
        i2s_buff = NULL;
    }
    vTaskDelete(NULL);
}
```

welkinchan, 2024-08-19:
> @welkinchan 根据你的现象， 像是使用 i2s 出现了问题， 如果使用了 i2s_stream 的话， 就建议使用 i2s stream 作为数据来源了，可以参考如下
>
> ```
> // First need initializsed i2s audio element as input
> static audio_element_handle_t i2s_reader;
> void [...]

很费劲得增加了i2s audio element组件之后，这个代码测试下来还是不行。了_了。

我把我的项目上传到这里了，如果有兴趣，请帮我看一下哈。https://github.com/welkinchan/chatgpt_wgdemo
我用i2s_stream的pipeline播放音频后，就无法再唤醒了。

closed 2025-02-24 by welkinchan