espressif/esp-sr #111: link issue on esp32s3

raptium, 2024-08-08:
I am using the esp-sr library in my Rust project and generating Rust bindings with the help of [esp-idf-template](https://github.com/esp-rs/esp-idf-template). Although my project compiles successfully, I'm encountering linking errors due to missing symbols that should be in libm/newlib.

```
Error: Linker /home/raptium/Pets/hello-esp/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/raptium/Pets/hello-esp/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/raptium/Pets/hello-esp/components/esp-sr/lib/esp32s3/libc_speech_features.a(c_speech_features.c.obj):(.literal.mfcc_create+0x28): undefined reference to `ceilf'
          /home/raptium/Pets/hello-esp/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/raptium/Pets/hello-esp/components/esp-sr/lib/esp32s3/libc_speech_features.a(c_speech_features.c.obj): in function `mfcc_create':
          /home/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/c_speech_features/c_speech_features.c:434:(.text.mfcc_create+0xf9): undefined reference to `ceilf'
          /home/raptium/Pets/hello-esp/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/raptium/Pets/hello-esp/components/esp-sr/lib/esp32s3/libc_speech_features.a(c_speech_features.c.obj): in function `csf_framesig':
          /home/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/c_speech_features/c_speech_features.c:1718:(.text.csf_framesig+0x3c): undefined reference to `ceilf'
          /home/raptium/Pets/hello-esp/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/raptium/Pets/hello-esp/components/esp-sr/lib/esp32s3/libc_speech_features.a(esp_speech_features.c.obj):(.literal.esp_win_func_init+0x44): undefined reference to `pow'

...
```

I suspect this is a similar issue that was previously fixed through commit [8ed7b711b33f963dff44dd5b619dc5673ece7955](https://github.com/espressif/esp-sr/commit/8ed7b711b33f963dff44dd5b619dc5673ece7955).

As a workaround, I added the following line to my project's CMakeLists.txt file:

```
add_prebuilt_library(c_speech_features "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libc_speech_features.a" PRIV_REQUIRES ${COMPONENT_NAME})
```

This seems to resolve the issue successfully.

Could you please confirm if this addition is necessary to fix the missing symbols problem?

feizi, 2024-08-09:
Thank you for your feedback. I'm not very clear on the library linking rules, but from experience, if a library cannot be found, it is necessary to use `add_prebuilt_library` in the CMakeLists to explicitly specify it.

closed 2024-10-08 by feizi

yaobo-lab, 2024-12-01:
@raptium  Finally, can you use this library in rust projects?

raptium, 2024-12-01:
> @raptium  Finally, can you use this library in rust projects?

Yes, with the workaroud I posted above. To be honest I don't understand what the cmake rule is for, but it works.