espressif/esp-sr #112: fail to load wn9 models on esp32-s3-korvo-1 board

raptium, 2024-08-08:
I am using esp-sr 1.7.1 and have also tried the latest master version. My project implements wake word detection using Wakenet models. Loading wn8 models (CONFIG_SR_WN_WN8_ALEXA=y) works fine, but when I switch to wn9 models (e.g., wn9_alexa, wn9_hilexin), my program constantly panics.

Could you please help me understand why my program panics when using wn9 models?

My hardware is an esp32-s3-korvo-1 dev board

```
I (956) AFE_SR: afe interface for speech recognition
I (956) AFE_SR: AFE version: SR_V220727
I (966) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1
I (976) AFE_SR: aec_init: 1, se_init: 1, vad_init: 1
I (976) AFE_SR: wakenet_init: 1

Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  0 register dump:
PC      : 0x4205691e  PS      : 0x00060130  A0      : 0x8205522d  A1      : 0x3fca33f0
A2      : 0x3fcb19c8  A3      : 0x3fcb21b0  A4      : 0xff031fb8  A5      : 0xffffffeb
A6      : 0x000000a4  A7      : 0x00000029  A8      : 0x3fc9e5f0  A9      : 0x3fcb20e4
A10     : 0x3fcb20bc  A11     : 0x00000000  A12     : 0x00000014  A13     : 0x3fcb21c4
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000005  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000

Backtrace: 0x4205691b:0x3fca33f0 0x4205522a:0x3fca3420 0x42048fd2:0x3fca3600 0x42005fec:0x3fca3650 0x420829bb:0x3fca3760 0x420051b8:0x3fca3780 0x4201a7c4:0x3fca37a0 0x4201a770:0x3fca37d0 0x4201a151:0x3fca3800 0x420051a8:0x3fca3830 0x42006263:0x3fca3860 0x4200908b:0x3fca3880 0x42089450:0x3fca38a0
```

```
0x4205691b: read_model_params at hufzip_model.c:155
0x4205522a: model_create at wakenet9_quantized.c:289
 (inlined by) model_init at wakenet9_quantized.c:884
0x42048fd2: afe_create_from_config at esp_afe_sr.c:334
0x42005fec: hello_esp::bindings::esp_sr::AudioFrontEnd::speach_recognition at esp_sr.rs:138
 (inlined by) hello_esp::app_main at main.rs:78
 (inlined by) hello_esp::main at main.rs:130
```

feizi, 2024-08-09:
Hi, @raptium ,
please try the [wake_word_detection example](https://github.com/espressif/esp-skainet/tree/master/examples/wake_word_detection) in esp-skainet.

raptium, 2024-08-10:
Thanks @feizi 

I've tried the example and it works well.

After spending half a day debugging, I have identified that the issue is caused by my use of `espflash write-bin` instead of `esptool` to flash `srmodels.bin`. I have yet to determine if I am using incorrect options for `espflash`, resulting in the corruption of the model data.

Considering that the issue does not pertain to esp-sr, I am closing it now.

closed 2024-08-10 by raptium