espressif/esp-sr #113: E (12869) SR_RINGBUF: Invalid size

Gyxqq, 2024-08-15:
I (12759) SR: FREE HEAP SIZE 106660
I (12759) SR: Model init success
I (12759) SR: AFE init success
I (12769) SR: FREE HEAP SIZE 106456
I (12769) AFE_SR: afe interface for speech recognition

I (12779) AFE_SR: AFE version: SR_V220727

W (12779) SR_SYS: PSRAM freq should be 80MHz
W (12789) SR_SYS: Flash freq should be 80MHz
W (12789) SR_SYS: Flash mode should be QIO
I (12799) AFE_SR: Initial auido front-end, total channel: 1, mic num: 0, ref num: 1

I (12809) AFE_SR: aec_init: 1, se_init: 1, vad_init: 1

I (12809) AFE_SR: wakenet_init: 1

Deprecated Quantized wakenet5: wakeNet5_v1_nihaoxiaozhi_6_0.983_0.95, mode:0 (Nov 21 2023 19:15:51)
Please switch to ESP32-S3 and wakenet9 !
E (12869) SR_RINGBUF: Invalid size
E (12869) SR_RINGBUF: Invalid size
I (12979) AFE_SR: wake num: 0, mode: 1, (Nov 21 2023 19:15:51)

I (12979) SR: FREE HEAP SIZE 45236
I (12979) SR: AFE create success
I (12979) SR: I2S BUFF MALLOC 1024
I (12979) main: init model success       (12989) SR: I2S READ DATA
I (12989) SR: FREE HEAP SIZE 27128
multinet:mn2_cn
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401eef92  PS      : 0x00060230  A0      : 0x800f4d32  A1      : 0x3fff9120
0x401eef92: sr_rb_bytes_available at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/sr_ringbuf.c:156

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0xfffffae6
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3fff02ac  A9      : 0x3fff90e0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
A14     : 0xbd234000  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff
0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x401eef8f:0x3fff9120 0x400f4d2f:0x3fff9140 0x400f5026:0x3fff9160 0x400d805c:0x3fff9180
0x401eef8f: sr_rb_bytes_available at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/sr_ringbuf.c:156
0x400f4d2f: afe_feed_aec_init_true at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr.c:502
0x400f5026: afe_feed at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr.c:619
0x400d805c: feed_Task at F:/projects/esp32_vase/main/speech_recognition.c:101

feizi, 2024-10-08:
please use esp32s3 or esp32p4.

closed 2024-10-08 by feizi

zggcd157, 2025-04-05:
请问你解决了吗

Gyxqq, 2025-04-10:
> 请问你解决了吗

换S系列

zggcd157, 2025-04-10:
> > 请问你解决了吗
> 
> 换 S 系列

我这边排除的问题是使用全部变量和开启PSRAM即可使用我是ESP32S3N32R8

Gyxqq, 2025-04-10:
> > > 请问你解决了吗
> > 
> > 换 S 系列
> 
> 我这边排除的问题是使用全部变量和开启PSRAM即可使用我是ESP32S3N32R8

我原来用32原版跑的，看来只能用S系了