espressif/esp-sr #118: srmodel_mmap_init fails when CONFIG_SPI_FLASH_ROM_IMPL is enabled

mike-2020, 2024-08-31:
After enable "CONFIG_SPI_FLASH_ROM_IMPL", failed to mmap model file. Looks like "spi_flash_mmap_get_free_pages" returns free heap size (2944KB is very likely the free heap size at the time) in such case. Disabling CONFIG_SPI_FLASH_ROM_IMPL will fix the issue.

`I (14340) MODEL_LOADER: The storage free size is 2944 KB
I (14340) MODEL_LOADER: The partition size is 4096 KB
E (14350) MODEL_LOADER: The storage free size of this board is less than model partition required size
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x42024d01
0x42024d01: srmodel_mmap_init at /home/xxx/esp-adf/components/esp-sr/src/model_path.c:272 (discriminator 1)

file: "/home/xxx/esp-adf/components/esp-sr/src/model_path.c" line 272
func: srmodel_mmap_init
expression: esp_partition_mmap(partition, 0, partition->size, ESP_PARTITION_MMAP_DATA, &root, models->mmap_handle)

abort() was called at PC 0x4038848b on core 0
0x4038848b: _esp_error_check_failed at /home/xxx/esp-idf-v5.2.2/components/esp_system/esp_err.c:50`

feizi, 2024-09-04:
Hi @mike-2020 ， why do you need to enable "CONFIG_SPI_FLASH_ROM_IMPL"?

mike-2020, 2024-09-05:
I was following IDF document to try to save IRAM usage. CONFIG_SPI_FLASH_ROM_IMPL is part of IDF doc'suggestion.

sun-xiangyu, 2024-09-06:
[This method does not seem to support Octal flash](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_flash/spi_flash_idf_vs_rom.html), but it is very important for esp-sr performance.

closed 2024-09-09 by mike-2020