espressif/esp-sr #119: How to save power when SR is running

mike-2020, 2024-09-09:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Does light sleep work with SR? Or is there a guideline for power saving while SR is running?

feizi, 2024-09-09:
The best way is to use low-power VAD.
 Currently, ESP32-P4 has a built-in low-power VAD on the LP coer. ESP32-S3 need to implement a low-power VAD on the CPU core. The VAD algorithm is simple and can be run at a lower frequency to reduce power consumption.

mike-2020, 2024-09-10:
For ESP32S3, the low power VAD is already implemented? Or it is going to be implemented?

feizi, 2024-09-11:
We do not plan to implement low power VAD for ESP32S3. There will be another solution in the future for esp32s3.

mike-2020, 2024-09-13:
ok. hope the new solution will be released soon.

But for now, when SR is in use, can I enable "light sleep" or "modem sleep"?