espressif/esp-sr #120: ESP-SR easily triggers watchdog

Sail-211010, 2024-09-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

E (12640891) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (12640891) task_wdt:  - IDLE0 (CPU 0)
E (12640891) task_wdt: Tasks currently running:
E (12640891) task_wdt: CPU 0: SR Feed Task
E (12640891) task_wdt: CPU 1: loopTask
E (12640891) task_wdt: Aborting.
E (12640891) task_wdt: Print CPU 0 (current core) backtrace

feizi, 2024-09-11:
ESP-SR requires real-time operation of multiple algorithms, which may trigger the watchdog. It is advised not to combine tasks such as data reading/writing and transmission with the feed or fetch tasks of ESP-SR.

We will also continue to optimize the resource consumption of ESP-SR.

Sail-211010, 2024-09-11:
Okay, I understand. Thank you.

feizi, 2024-09-11:
If you need to save or transmit data, I recommend using a ring buffer as an intermediate cache to separate the data transfer tasks from the ESP-SR tasks. This way, you can set a larger ring buffer to reduce task blocking and potential data loss.

Sail-211010, 2024-09-11:
How do I need to achieve this ”ring buffer as an intermediate cache“

feizi, 2024-09-12:
Please refer to [esp_ringbuf ](https://github.com/espressif/esp-idf/tree/59e18382702b6986be3d3f55e9ac7763c1397cf7/components/esp_ringbuf)

Sail-211010, 2024-09-12:
Although this may be a good solution, it may be a little difficult for me. According to the existing program, I do not know where and how to modify it, even though you have provided me with a reference example path
Thank you so much for your answer!

closed 2024-10-08 by feizi