espressif/esp-sr #122: AEC scaling

boborjan2, 2024-11-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

It would be nice to be able to modify major parameters of AEC (e.g. filter length) so that in use cases where it is enough to suppress shorter echo only, AEC could work using less memory.

### Use cases

Lower memory footprint.

boborjan2, 2024-11-20:
Btw which webrtc AEC implementation do you use? AEC3 or AECM?

sun-xiangyu, 2024-11-21:
Hi @boborjan2 , we do not use webrtc AEC.

boborjan2, 2024-11-22:
I see. The library contains several webrtc symbols along with speex ones, you might want to rename them then.

boborjan2, 2024-12-08:
A question regarding noise suppression: are the noise suppression functions intended to use stand-alone? I see that there are an example showing that VAD can be used without AFE. Is this the case with NS as well? Thanks, Viktor.

closed 2024-12-19 by boborjan2